qiskit-community/qiskit-metal #420: Fix linting

JeremyDrysdale-ibm, 2021-03-18:
Please be as descriptive as possible, including: what is expected, why is this feature needed, what is the objective, etc.

### What is the feature being requested?
Fix linting configuration to allow linting to pass and ignore the long lines we want to keep long

priti-ashvin-shah-ibm, 2021-03-19:
Some FYI information.
To run pylint, on the command line at top of qiskit-metal.  

`$ pylint qiskit_metal | tee  hold_output.txt`

Metal repo has the .pylint file at
qiskit-metal/.pylint

A good file to compare Metal's pylint to is the pylint with Qiskit Terra.
https://github.com/Qiskit/qiskit-terra/blob/master/.pylintrc

priti-ashvin-shah-ibm, 2021-03-19:
There is a branch called "linting"

JeremyDrysdale-ibm, 2021-03-19:
Fixed linting errors in the test module.

priti-ashvin-shah-ibm, 2023-02-01:
Everything has not been linted, but we have been linting as merging has been happening.

closed 2023-02-01 by priti-ashvin-shah-ibm