qiskit-community/qiskit-metal #430: qiskit_metal can not be imported due to METALGUI error

MAN6O, 2021-03-19:
### Information

- **Qiskit Metal version**: 0.003
- **Python version**: 3.7.8
- **Operating system**: Windows 10

### What is the current behavior?
After following the installation guide step by step, I tried to run the first tutorial script.
When importing qiskit_metal the following error message is displayed in jupyter-notebook:

~\Anaconda3\envs\my-env\lib\site-packages\pyqode\core\dialogs\goto.py in <module>
      6 
      7 
----> 8 class DlgGotoLine(QtWidgets.QDialog, dlg_goto_line_ui.Ui_Dialog):
      9     """
     10     Goto line dialog.

AttributeError: module 'pyqode.qt.QtWidgets' has no attribute 'QDialog'

### Steps to reproduce the problem
Import qiskit_metal 

### What is the expected behavior?
To import qiskit_metal without errors

### Suggested solutions
perhaps the wrong version of pyqode is installed ? My version is pyqode.python 2.12.1

marcolincs, 2021-03-19:
This is a known issue, that arises when in the virtual environment there is both installation of the production pyqode.qt and the github forked pyqode.qt.
Can you please verify with `pip freeze` that you are using the git version? Should looks something like this:
[image]

MAN6O, 2021-03-19:
Thanks for the quick reply. I was able to solve the problem:
During the installation with conda due to an internal access error the wrong pyqode.qt version got installed. I had to manually resolve it, now it works.

Best wishes

closed 2021-03-19 by MAN6O