qiskit-community/qiskit-metal #432: ModuleNotFoundError: No module named 'qiskit_metal._gui.widgets.build_history'

leo07010, 2021-03-20:
### Information

- **Qiskit Metal version**:
- **Python version**:
- **Operating system**:

### What is the current behavior?

ModuleNotFoundError: No module named 'qiskit_metal._gui.widgets.build_history'

### Steps to reproduce the problem
import qiskit_metal as metal

### What is the expected behavior?
It should pass

### Suggested solutions
rename the 'qiskit_metal._gui.widgets.build_history'

dmtvanzanten, 2021-03-20:
Don't know whether this is useful to anyone. 
I found the missing module in the commit [a800be089f](https://github.com/Qiskit/qiskit-metal/tree/a800be089f83579eea9fb8b85c60c5b38f152de0/qiskit_metal/_gui/widgets/build_history).
Creating a folder 'qiskit_metal/_gui/widgets/build_history' and coping the content seemed to have solved the issue for me.

marcolincs, 2021-03-22:
hi @leo07010 ,
can you please fill up the top part of the bug form? I cannot replicate the issue on my machine and need that information to try anything else.

leo07010, 2021-03-22:
[image]

marcolincs, 2021-03-22:
@leo07010 , I believ eyou see it, But unless I reproduce it and see the same error, I cannot help you.
Please provide the info at the top of the form (OS, library and python version)

leo07010, 2021-03-22:
Oh，OK

marcolincs, 2021-03-22:
Alright, so it slooks like you and I are both on windows. That is good.
This is a problem: python 3.8.5.
if you had installed qiskit-metal using the environment.yml, you would be seeing 3.7.8. How come you see 3.8.5? Can you recreate a new conda environment using the environment.yml?

leo07010, 2021-03-22:
yes I can 
I successfully solved this problem, I think we can’t use powershell to install the package, but use anaconda prompt

leo07010, 2021-03-22:
Because I did exactly the same thing in the prompt and successfully executed it

marcolincs, 2021-03-22:
I use Powershell.
-> jupyter notebook or jupyter lab, default kernel is `base` if launched from a Powershell that is not admin. if you are running as admin the powershell, then jupyter notebook will default to running ion presently active conda environment.
-> there are instructions on the README to create a new kernel for the specific conda environment so that you can select it in jupyter notebook and jupyter lab.

By executing inside jupyter notebook/lab like you did, you are fundamentally installing things in whatever is the active environment, which is likely the base. So that might cause some issue....unless you are actually creating and activating a new conda env from inside jupyter....that I did not test, but I can see how that might potentially work, but I wonder what would happen if you reboot.

Anyways, I hope this note helps in case you find further problems that might be resulting from installing things that way.
Closing the issue for now

closed 2021-03-22 by marcolincs