qiskit-community/qiskit-metal #433: ImportError: attempted relative import with no known parent package

leo07010, 2021-03-20:
### Information

- **Qiskit Metal version**:
- **Python version**:
- **Operating system**:

### What is the current behavior?

---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
<ipython-input-17-4fecc404dbe1> in <module>
----> 1 from .widgets.all_components.table_model_all_components import \
      2     QTableModel_AllComponents

ImportError: attempted relative import with no known parent package

### Steps to reproduce the problem

from .widgets.all_components.table_model_all_components import \
    QTableModel_AllComponents

### What is the expected behavior?

It should pass

### Suggested solutions

giving new file

marcolincs, 2021-03-22:
@leo07010 , did this case also resolve with the #432 change?
if not, where are you writing the two lines mentioned? We would need to understand how to reproduce it.
....just writing the line in jupyter notebook is expected to fail (python makes relative referencing only work within the package, not from jupyter notebook)

leo07010, 2021-03-22:
yes,it was also resolved
Thanks

closed 2021-03-22 by marcolincs