qiskit-community/qiskit-metal #441: Adding the option to invert the gds file when exporting the design

MAN6O, 2021-03-25:
Please be as descriptive as possible, including: what is expected, why is this feature needed, what is the objective, etc.

### What is the feature being requested?
The option to invert the polygons in the gds chip design, so that the polygons do not describe the metal structures, but the regions of the sc material that needs to be removed during fabrication. 
It can be a checkbox  ([ ] invert design) in the gui export window.

### What are use cases for this feature?
Exporting gds designs and having them inverted saves users the time to edit the code, or do it manually in a GDS-editor.

zlatko-minev, 2021-04-01:
We should definitely do this. @priti-ashvin-shah-ibm  can help you also if you want to give it a first try shot

priti-ashvin-shah-ibm, 2021-05-03:
This is being addressed.  In particular - allow for negative mask by selecting the chip and layer.

closed 2021-05-04 by JeremyDrysdale-ibm