qiskit-community/qiskit-metal #447: Add the option to include ground between coupler pads and qubits

MAN6O, 2021-03-30:
Please be as descriptive as possible, including: what is expected, why is this feature needed, what is the objective, etc.

### What is the feature being requested?
Sometimes it is useful to have the option to include ground between the couplers of qubits and the qubits, right now one has to change the charge line parameters to get something like this which is a restriction.

### What are use cases for this feature?
The ability to create more unique qubit setups and architectures without breaking grounds.

zlatko-minev, 2021-04-01:
Is there a sketch of what you have in mind? It sounds to me a little like this?
[image]

from https://qiskit.org/documentation/metal/circuit-examples/index.html

MAN6O, 2021-04-15:
No, I actually mean it like this:

[image]

So that there is ground between the coupling pads and the qubit.

priti-ashvin-shah-ibm, 2023-02-01:
Please make your own custom component.

closed 2023-02-01 by priti-ashvin-shah-ibm