qiskit-community/qiskit-metal #475: Add video playlist to docs

zlatko-minev, 2021-04-15:
### What is the documentation modification?
 Add video page and link in TOC like this 
[image]

to the video playlist 

https://www.youtube.com/watch?v=bsrsKZLTkTo&list=PLOFEBzvs-VvqHl5ZqVmhB_FcSqmLufsjb&index=2&ab_channel=Qiskit 

Can embed playlist or maybe better yet, link to playlist but also listinvidiaully each video with embed, see 

* https://techwritingmatters.com/documenting-with-sphinx-tutorial-part-3-writing
* https://docs.typo3.org/m/typo3/docs-how-to-document/master/en-us/WritingReST/YoutubeVideos.html

marcolincs, 2021-04-15:
good idea.

JeremyDrysdale-ibm, 2021-04-26:
It seems to me that a link to the playlist is sufficient over embedding individual videos at this time since the playlist is updating as new videos are being made and published.

I suggest
- Link to the YouTube playlist now
- Add a page to the docs with embedded videos when we have completed the current round of video tutorial sessions

Otherwise, we are stuck updating the docs to embed new videos as each one is published.

PR #496 (2021-04-26): Added tutorial video link to the docs

### What are the issues this pull addresses (issue numbers / links)?
Closes #475 

### Did you add tests to cover your changes (yes/no)?
Not necessary

### Did you update the documentation accordingly (yes/no)?
Yes

### Did you read the CONTRIBUTING document (yes/no)?
Yes

### Summary
Added tutorial video link to the docs, and link in the README.
closed 2021-04-26 by JeremyDrysdale-ibm