qiskit-community/qiskit-metal #481: BBQ Analysis based on EPR

dwang429, 2021-04-19:
Please be as descriptive as possible, including: what is expected, why is this feature needed, what is the objective, etc.

### What is the feature being requested?
Black box quantization techniques that build on existing EPR code

### What are use cases for this feature?
Quantum circuit analysis for driven modal simulations

dwang429, 2021-10-01:
Based on Grace's and my discussion with Hanhee, her old paper (https://arxiv.org/pdf/1204.0587.pdf) is outdated and the new BBQ analysis based on Firat's paper ought to be used instead.

SamarthH, 2022-01-14:
Is it not better to implement the newest one by Solgun, DiVincenzo, and Gambetta https://ieeexplore.ieee.org/document/8633444 ?

ThomasGM4, 2022-01-14:
> Is it not better to implement the newest one by Solgun, DiVincenzo, and Gambetta https://ieeexplore.ieee.org/document/8633444 ?

Yep, that is now the plan. We just haven't had time to start working on the new analysis code for it, but we do have it in mind. If anyone wanted to start trying to take a stab at it I would certainly encourage that!

priti-ashvin-shah-ibm, 2023-02-01:
We encourage the community to contribute the BBQ stand-alone analysis of simulation data.