qiskit-community/qiskit-metal #491: Update for 3D structure layer stack

JeremyDrysdale-ibm, 2021-04-22:
Please be as descriptive as possible, including: what is expected, why is this feature needed, what is the objective, etc.

### What is the feature being requested?
Update metal to allow for 3D structure layer stack.

After investigating what is needed, please made addition issues as needed in support of this issue.

sandokokosen, 2021-06-15:
Hi, I'm currently looking into the use of qiskit-metal for design and simulation of 3D systems. Is this being actively worked on at the moment by the metal team, or has been postponed to later dates? I have made some strides into this but I notice that, for instance most of the codes that I can get hold of are still assuming a single chip called 'main', and the chip should be drawn from the z=0 origin in ansys. I am actively editing renderer_ansys.py at the moment just to see how far I can imitate my usual workflow in ansys, but I have been wondering if I will at some point stray too far away from the intended structure.

zlatko-minev, 2021-08-10:
@sandokokosen  were you able to make progress on this?

sandokokosen, 2021-08-10:
Hi @zlatko-minev , I think so. I have been able to render the flip-chip stack in ansys using metal and let it do the things I would like it to do. Although I did notice that the team has enforced all chipname as 'main' in the latest code I got, any other name will give error as far as I remember. So I have disable that function in my case, because I'd like to have the chips to have different name in ansys, and when they are rendered later for gds file output.

zlatko-minev, 2021-08-11:
The chip naming thing should be temporary (maybe @dwang429 recalls?). If you make a pull request with your update it will be easier to see.

sandokokosen, 2021-08-11:
Sorry I was not being precise enough with what I wrote, I was specifically looking into flip-chip layer stacks when I meant 3D systems, not 3D cavity system. Should the pull request still be made?

zlatko-minev, 2021-08-11:
Open source community grows and thrives on contributions :)