qiskit-community/qiskit-metal #498: QComponent pin options update

marcolincs, 2021-04-28:
### Information

- **Qiskit Metal version**: 0.0.3
- **Python version**: 3.7.8
- **Operating system**: Windows

### What is the current behavior?
[image]

### Steps to reproduce the problem
try to options.update() giving a new pin to the options

### What is the expected behavior?
should allow to update those options (thus create a new pin) after the component was created

priti-ashvin-shah-ibm, 2021-05-04:
Am guessing that  `design.overwrite_enabled = True` is in your notebook.  When one re-runs a cell, the component is deleted from QDesign and new one is generated.  So the name of component could be changed, if the names are auto-generated.  Am guessing that you re-ran the cell earlier.  Then the name of the component has been changed. 

To better understand the issue, need to know what was before this and if you re-ran any cells.  

q1.options is probably pointing to the previous qcomponent.

marcolincs, 2021-05-04:
Of course....,
yes, that line is there, but does not matter here because I am not re-running the cell, I am running options.update(). You can recreate by creating the component with X pins and then trying to add one additional pin.