qiskit-community/qiskit-metal #527: Buttons not visible

manannarang, 2021-05-18:
### Information

- **Qiskit Metal version**:
- **Python version**:
- **Operating system**:

### What is the current behavior?
Unable to see the buttons below

### Steps to reproduce the problem

Press Q3

### What is the expected behavior?

Buttons should be clearly visible when I minimize or maximize the window.

### Suggested solutions

Add a scrollbar to the right

JeremyDrysdale-ibm, 2021-05-18:
Hi @manannarang,

Please provide us with a bit more detail.  You mention in the text that you are unable to see the button "below", but there is no further information and/or screenshot to help us reproduce this bug.

manannarang, 2021-05-18:
[image]

priti-ashvin-shah-ibm, 2021-05-24:
We can add a scroll bar for GDS, HFSS, and Q3D renderers.

marcolincs, 2021-07-07:
The correct way to fix this is to make the two widgets resizable with the window, rather then current fixed size.

Indeed, the widgets already have embedded scroll-bar, so making them resizable will automatically utilize their scroll-bar as needed.....and in most cases it will not be needed because making the widget resizable will pull up the bottom buttons to as high up as possible.

it might be slightly more complicated than that in practice, but the gist is: we need to fix the widgets, not the window. Cheers.

zlatko-minev, 2021-08-10:
@grace-harper-ibm , did we tackle this one yet?