qiskit-community/qiskit-metal #548: Add any missing components to the Circuit Library docs

zlatko-minev, 2021-06-01:
### What is the documentation modification?

Add any missing components to the Circuit Library docs

[image]

such as transmon_pocket_cl is not in there. Let's add all the available components 
[image]

zlatko-minev, 2021-06-01:
@JeremyDrysdale-ibm   perhaps for Sagarika when she add her new qubit design or a good test project for the interns, @ThomasGM4  @marcolincs  to create and test all the components and add them in here?
[image]

ThomasGM4, 2021-06-01:
Do we want all components covered, just primary parents (eg. transmon with flux line another separate or just transmon pocket), or just basic subset of common components used?

PR #661 (2021-08-03): Add additional circuit examples

### What are the issues this pull addresses (issue numbers / links)?
Closes #548 

### Did you add tests to cover your changes (yes/no)?
No

### Did you update the documentation accordingly (yes/no)?
Yes

### Did you read the CONTRIBUTING document (yes/no)?
Yes

### Summary
Added additional circuit examples
closed 2021-08-03 by JeremyDrysdale-ibm