qiskit-community/qiskit-metal #549: Small bug in layout of the GDS/Q3D export window in the gui

zlatko-minev, 2021-06-01:
### Information

### What is the current behavior?
Small bug in layout of the GDS/Q3D export window in the gui when you rescale it 
[image]

### Steps to reproduce the problem

Try to rescale the window 

### What is the expected behavior?

The window should rescale the contnes inside 

### Suggested solutions

use a `layout` container

zlatko-minev, 2021-06-01:
@JeremyDrysdale-ibm  someone who knows the gui can help tackle this, shouldn't be too hard. Mostly add a layout container I think to the way it is laid out.

marcolincs, 2021-09-10:
Connecting issues
Duplicates #527 , yet offers an easy way to test things on MAC OS