qiskit-community/qiskit-metal #553: When notebooks are re-run on same server, with Ansys, we get different output.

priti-ashvin-shah-ibm, 2021-06-02:
### Information

- **Qiskit Metal version**:
- **Python version**:
- **Operating system**:

### What is the current behavior?
During review of pr, noticed the Ansys output is different when run on the same server. 

### Steps to reproduce the problem
https://github.com/Qiskit/qiskit-metal/pull/552/commits/5bf4e6e62f8cb4f25655cca768676511c1ac26c7
Don't know how, but this commit shows the difference. 

### What is the expected behavior?

Expect Ansys to give same results.

### Suggested solutions

priti-ashvin-shah-ibm, 2021-06-02:
https://github.com/Qiskit/qiskit-metal/pull/552/commits/5bf4e6e62f8cb4f25655cca768676511c1ac26c7

For pull request #552

JeremyDrysdale-ibm, 2021-07-08:
The shot below shows the comparison of the results after running the south korea hackathon tutorial.  The differences are shown a little clearer here.  On the left is the output after the first run, the on the right is the output after the second run.
[image]

JeremyDrysdale-ibm, 2021-07-15:
The first few passes won't be too useful.  Line 8 above is where we should focus, they clearly converge.  These numbers are "close enough".

closed 2021-07-15 by JeremyDrysdale-ibm