qiskit-community/qiskit-metal #562: Delete button not working

manannarang, 2021-06-05:
### Information

- **Qiskit Metal version**:
- **Python version**:
- **Operating system**:

### What is the current behavior?

Cannot delete component via button

### Steps to reproduce the problem

Ref Video

### What is the expected behavior?

Component should delete

### Suggested solutions

Remove component from list

https://user-images.githubusercontent.com/52881942/120905608-d65d2f00-c670-11eb-8ae3-c4ba69828366.mp4

zlatko-minev, 2021-08-10:
@JeremyDrysdale-ibm , this looks like a simple fix of the action assigned to the gui button, can we see to fix this?

JeremyDrysdale-ibm, 2021-08-11:
It appears that this bug was fixed but the issue never closed.

@manannarang If you continue to experience this issue a current pull of metal code, please reopen the issue.

closed 2021-08-11 by JeremyDrysdale-ibm

zlatko-minev, 2021-08-11:
Great, thanks￼