qiskit-community/qiskit-metal #570: Installation error of Qiskit metal.

RamAIbot, 2021-06-10:
### Information

- **Qiskit Metal version**:0.0.3
- **Python version**:3.9.4
- **Operating system**:Windows

### What is the current behavior?
After installing Qiskit_metal in a separate conda environment while trying to import it the below error occurs.
```
>>> import qiskit_metal
PySide2/__init__.py: Unable to import shiboken2 from ,
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "F:\error\qiskit-metal\qiskit_metal\__init__.py", line 113, in <module>
    __setup_Qt_backend()
  File "F:\error\qiskit-metal\qiskit_metal\__init__.py", line 57, in __setup_Qt_backend
    from PySide2 import QtCore  #, QtWidgets
  File "D:\Softwares\AnacondaFull\envs\metalerror\lib\site-packages\PySide2\__init__.py", line 107, in <module>
    _setupQtDirectories()
  File "D:\Softwares\AnacondaFull\envs\metalerror\lib\site-packages\PySide2\__init__.py", line 58, in _setupQtDirectories
    import shiboken2
  File "D:\Softwares\AnacondaFull\envs\metalerror\lib\site-packages\shiboken2\__init__.py", line 30, in <module>
    _init_pyside_extension()
NameError: name '_init_pyside_extension' is not defined
```

### Steps to reproduce the problem

```
conda env create -n <env_name> environment.yml
conda activate <env_name>
python -m pip install -ve .
python
import qiskit_metal

```
### What is the expected behavior?

### Suggested solutions

manannarang, 2021-06-10:
I would request you to once try to open jupyter notebook and see if it is working. Kindly write the following commands in Anaconda prompt or preferably Miniconda prompt:

1.) conda env create -n <env_name> environment.yml
2.) conda activate <env_name>
3.) python -m pip install -ve .
4.) cd C:\Users\manan_narang\Documents\GitHub\qiskit-metal
5.) jupyter notebook
6.) Open Tutorials Folder and try to run the .ipynb file

Another alternate would be to try and open it in VSCode by activating the kernel in it and running the files.

Hope this helps!

ambhudia, 2021-06-11:
I am experiencing the same problem on Ubuntu 20.04 with the same version of Python and Metal. 
Both `PySide2-5.15.2` and `PySide2-5.15.1` yield the same error.

```PySide2/__init__.py: Unable to import shiboken2 from [REDACTED]

---------------------------------------------------------------------------
NameError                                 Traceback (most recent call last)
<ipython-input-1-b408c55ac936> in <module>
----> 1 import qiskit_metal as metal

~/qiskit-metal/qiskit_metal/__init__.py in <module>
    111 
    112 
--> 113 __setup_Qt_backend()
    114 del __setup_Qt_backend
    115 

~/qiskit-metal/qiskit_metal/__init__.py in __setup_Qt_backend()
     55     os.environ["QT_API"] = "pyside2"
     56 
---> 57     from PySide2 import QtCore  #, QtWidgets
     58     from PySide2.QtCore import Qt
     59 

~/anaconda3/envs/metal/lib/python3.9/site-packages/PySide2/__init__.py in <module>
    105                 os.environ['PATH'] = path
    106 
--> 107 _setupQtDirectories()

~/anaconda3/envs/metal/lib/python3.9/site-packages/PySide2/__init__.py in _setupQtDirectories()
     56 
     57     try:
---> 58         import shiboken2
     59     except Exception:
     60         paths = ', '.join(sys.path)

~/anaconda3/envs/metal/lib/python3.9/site-packages/shiboken2/__init__.py in <module>
     28 
     29 # Trigger signature initialization via __builtins__.
---> 30 _init_pyside_extension()

NameError: name '_init_pyside_extension' is not defined

```

RamAIbot, 2021-06-12:
Same error even in jupyter notebook.

```
---------------------------------------------------------------------------
NameError                                 Traceback (most recent call last)
<ipython-input-2-5b6c2e9a8ebe> in <module>
----> 1 import qiskit_metal as metal
      2 from qiskit_metal import designs, draw
      3 from qiskit_metal import MetalGUI, Dict, open_docs
      4 
      5 get_ipython().run_line_magic('metal_heading', 'Welcome to Qiskit Metal!')

f:\error\qiskit-metal\qiskit_metal\__init__.py in <module>
    111 
    112 
--> 113 __setup_Qt_backend()
    114 del __setup_Qt_backend
    115 

f:\error\qiskit-metal\qiskit_metal\__init__.py in __setup_Qt_backend()
     55     os.environ["QT_API"] = "pyside2"
     56 
---> 57     from PySide2 import QtCore  #, QtWidgets
     58     from PySide2.QtCore import Qt
     59 

D:\Softwares\AnacondaFull\envs\metalerror\lib\site-packages\PySide2\__init__.py in <module>
    105                 os.environ['PATH'] = path
    106 
--> 107 _setupQtDirectories()

D:\Softwares\AnacondaFull\envs\metalerror\lib\site-packages\PySide2\__init__.py in _setupQtDirectories()
     56 
     57     try:
---> 58         import shiboken2
     59     except Exception:
     60         paths = ', '.join(sys.path)

D:\Softwares\AnacondaFull\envs\metalerror\lib\site-packages\shiboken2\__init__.py in <module>
     28 
     29 # Trigger signature initialization via __builtins__.
---> 30 _init_pyside_extension()

NameError: name '_init_pyside_extension' is not defined

```

[image]

manannarang, 2021-06-12:
I think the problem lies with the installation. I would suggest you re-install qiskit metal. I faced some issues while installing and it took me three tries to get it up and running. If it is possible, try to install it in miniconda as it is much simpler.

marcolincs, 2021-07-07:
Hi,
I did not notice this case. This issue is addressed in the FAQ. it is a known issue due to "upgrading" PySide2 to a newer version (that goes with the newest python). Creating a new environment will sometime not help because the environment picks on shared libraries or residuates that still point to the other API.

closed 2021-07-07 by marcolincs