qiskit-community/qiskit-metal #578: Handling Ansys path_vars

warrench, 2021-06-15:
### Information

- **Qiskit Metal version**: 0.0.3
- **Python version**:
- **Operating system**:

### What is the current behavior?

Ansys renderer defaults to 2020 R2 and gets overridden on updates to the repository if one has a different installation.

See open_ansys() in ansys_renderer.py

### Steps to reproduce the problem

### What is the expected behavior?

USER should be able to specify a config file that holds their path variable which should be in the .gitignore so that updates to the repo do not overwrite when syncing

### Suggested solutions

Create a config json, or .yaml or some kind of easily readable config file which holds certain path information which can be looked up on execution for the default keyword rather than hardcoding the keyword to some default version of Ansys. This file can be placed in the .gitignore so that it is not changed on updating the repository. A function should be created which generates this file with some default values on setup of the repository. These can then be overwritten by the user at any point.

zlatko-minev, 2021-08-10:
@JeremyDrysdale-ibm , could we take a look at taking care of this?

marcolincs, 2021-09-10:
Hi @warrench, with the QAnalysis class approach (latest main branch), we are abandoning the open_ansys() command and use directly the COM, which does not require to specify the path as it will default to the most recent ansys installation (as far as we can tell).....in other words, the path is kept by the windows registry.

Does this address fully the needd of this issue?

priti-ashvin-shah-ibm, 2023-02-01:
Changing to pyaedt renderer for Ansys, so will close this.

closed 2023-02-01 by priti-ashvin-shah-ibm