qiskit-community/qiskit-metal #590: Checkbox not visible

manannarang, 2021-06-17:
### Information

- **Qiskit Metal version**: Latest
- **Python version**: Latest
- **Operating system**: Windows 10

### What is the current behavior?

Cannot check the components to render

### Steps to reproduce the problem

Open HFSS or Q3D

### What is the expected behavior?

[image]

### Suggested solutions

Add the checkbox

JeremyDrysdale-ibm, 2021-06-18:
Workaround until the bug is fixed: Switch your `Color theme` in the `Window View` dropdown to `System default` and the checkboxes will display.

JeremyDrysdale-ibm, 2021-06-18:
See https://github.com/Qiskit/qiskit-metal/issues/244

manannarang, 2021-06-18:
This is interesting

[image]

marcolincs, 2021-06-18:
Any help solving this is welcome. After spending quite some time on it we had to lower priority of #244 and get on other items.

manannarang, 2021-06-19:
I tried looking up in the code but could not find anything.

zlatko-minev, 2021-08-10:
This is probably an issue with the layout QT style file I defined that is loaded. There may be a missing image or the color of the checkbox is not set right. Does anyone else have this issue in dark?

priti-ashvin-shah-ibm, 2023-02-01:
I don't see this issue.  Having Qdarkstyle is in dependencies may have fixed this as part of other work.