qiskit-community/qiskit-metal #594: Keep cursor in place when editing Param Entry Window

grace-harper-ibm, 2021-06-18:
Please be as descriptive as possible, including: what is expected, why is this feature needed, what is the objective, etc.

### What is the feature being requested?
Currently, when you enter an argument into Param Entry Window, the cursor gets moved to the top of the window. 
Instead, keep the cursor wherever it last was

### What are use cases for this feature?
Ease of use

zlatko-minev, 2021-09-29:
To clarify a bit, the issue is when you open the dialog to create a new component, you see this 
[image]

Then if you open the options and edit the name of some key, the focus of the cursor is lost and reset. 

Ideally, the cursor and highlight would stay where it is. This one is a minor feature update

priti-ashvin-shah-ibm, 2023-02-01:
Looks like this has merged by @ismirlia , will close.

closed 2023-02-01 by priti-ashvin-shah-ibm