qiskit-community/qiskit-metal #624: GUI is not working under VS Code or terminal

anadb, 2021-06-27:
Qiskit-metal v. 0.0.3
Python v. 3.9.5
Linux Mint 18.3

I am trying to run GUI from vs code or just a terminal with simple code

`import qiskit_metal as metal
from qiskit_metal import designs, draw
from qiskit_metal import MetalGUI, Dict, open_docs

design = designs.DesignPlanar()
gui = MetalGUI(design)`

The window is opening but it is frozen. When I am running from jupyter it works ok.

priti-ashvin-shah-ibm, 2021-06-28:
@anadb , I think it has to do with starting GUI from within vscode.  This has been copied from slack workspace named `qiskit.slack.com`  , in particular channel #metal.   

You need to add  gui.qApp.exec_()   within your script at spot in code where you want the focus to switch to GUI.  

https://qiskit.slack.com/archives/C01R8KP5WP7/p1621266737040800?thread_ts=1621259670.039500&cid=C01R8KP5WP7

priti-ashvin-shah-ibm, 2021-06-28:
Regarding debugger for Jupyter lab , there is an improved debugger, but not have played with it yet.:
https://blog.jupyter.org/jupyterlab-3-0-is-out-4f58385e25bb

I use vscode.  I open the notebook in vscode, export the notebook to python script.  Use the right button. in image below. 
https://files.slack.com/files-pri/T7RSPHKK2-F01SV82B430/image.png

Then remove the lines that have:
%% [markdown]
when using find/replace `%% \[markdown\]`
and
%%
and
comment out the lines that have
get_ipython()
Then I can run the notebook in vscode and am able to set breakpoints.
If you want to step through pyEPR, you need to make separate editable clone, so, not to use the binary.  Then I changed the path to go to editable pyEPR before binary.
Not sure if this is what you are looking for. 
Also, when I work with vscode, I can’t interact with the Metal GUI.  If I want to change the focus to gui, I have to put into my script:  gui.qApp.exec_()   Don’t skip the underscore after letter c. (edited) 

[image]

anadb, 2021-06-28:
gui.qApp.exec_() resolved the problem.

closed 2021-06-28 by anadb