qiskit-community/qiskit-metal #625: GDS - more documentation for layers and their definitions

priti-ashvin-shah-ibm, 2021-06-28:
Please be as descriptive as possible, including: what is expected, why is this feature needed, what is the objective, etc.

### What is the feature being requested?
Even thought the content in  layers has been fluid, and may get reorganized, it would be good to have a place to get the latest information.  

### What are use cases for this feature?

marcolincs, 2021-06-29:
Hi,

I see this request comes from the slack channel. I believe this can be closed.

Layers are *arbitrary* pairing of number pairs with data-prep or manufacturing intention. These cannot be standardized (given a generic definition), because they are strictly related to any one handshake with their manufacturing vendors, or internal data-prep teams. That means that documenting/standardizing them would be a totally *arbitrary* exercise.

I suppose one could have a `for-students mock` definition stashed somewhere, but I advise to consider that only after completion of the queued up work on layer support.

priti-ashvin-shah-ibm, 2021-06-29:
@marcolincs , I think the focus of this issue is not to define the layers for Metal.  It is just to explain he layer numbers which came from export_to_gds() method.

marcolincs, 2021-06-29:
`numbers` is what I am referring to

marcolincs, 2021-06-29:
More copy-paste from the Slack channel (message from me to Siamack):

Priti pointed to the correct [notebook](https://github.com/Qiskit/qiskit-metal/blob/main/tutorials/3%20Renderers/3.2%20Export%20your%20design%20to%20GDS.ipynb). It has an example of how to change the layer number (first integer of the layer), check the cell with a few `cpw1.options.layer = 14` in it. At this time, this needs to be done per component. For the datatype (second integer of the layer) you can change some of them in the gds renderer options (see cell with a_gds.options). A couple of the datatypes are hard-coded at this time and will be made more accessible later.
Finally the junction layers that you see in the output come from the imported GDS of the junctions, which is only a dummy layout, thus should not be truly understood, but rather replaced with any real junction layout you might have.

In this issue we need:
- [ ] clarifying document describing the current layer defaults that show up in the output
- [ ] a tutorial notebook that focusses on how to change them
Could have both of the above in the same notebook.

closed 2021-07-16 by JeremyDrysdale-ibm