qiskit-community/qiskit-metal #631: Separate renderers from design

marcolincs, 2021-07-02:
Currently renderers are created when a design object is created and they are hosted inside the design object.
[image]
However we cannot assume design (or GUI) will be there for analysis/simulation (for example: the user might load a design to the simulator from a gds file). Therefore renderers must be accessible in a different way.

Different ways to solve this, open to suggestions:
- Suggestion 1. if we want to keep the renderer objects centralized (instantiated once), then we need to create them outside the design (and globally accessible), could introduce the concept of "app" or "desktop" wrapping it all.
- Suggestion 2. if we are ok decentralizing, then a new renderer object can be created for each analysis instance.

priti-ashvin-shah-ibm, 2023-02-01:
This works, if we have agreed to a scenario to import gds files.

priti-ashvin-shah-ibm, 2023-02-01:
It is much further down the time line.