qiskit-community/qiskit-metal #638: Critical errors popping up while trying to run Metal GUI

ani-tifr, 2021-07-13:
Information

- Qiskit Metal version: 0.0.4 
- Python version 3.7.10
- Operating system: Windows 10

While running gui = MetalGUI(design) syntax to open the GUI, I am getting the following errors:

_**05:10PM 17s CRITICAL [_qt_message_handler]: line: 0, func: None(), file: None  WARNING: QObject::moveToThread: Current thread (0x241ec5ea9b0) is not the object's thread (0x241dfce4430).
Cannot move to target thread (0x241ec5ea9b0)

05:10PM 17s CRITICAL [_qt_message_handler]: line: 0, func: None(), file: None  INFO: Could not load the Qt platform plugin "windows" in "" even though it was found.

Restarting kernel...**_
 
Please suggest some solutions. 

Using PyQt version 5.12.3

JeremyDrysdale-ibm, 2021-07-13:
Please provide us with a little more detail.  What python code was executed _prior_ to the `gui = MetalGUI(design)` call?

ani-tifr, 2021-07-13:
I started with the initial python code for Qiskit metal:

>>> from qiskit_metal import designs, draw, MetalGUI, Dict, open_docs
>>> design = designs.DesignPlanar()
>>> design.overwrite_enabled = True
>>> design.chips.main
>>> design.chips.main.size.size_x = '11mm'
>>> design.chips.main.size.size_y = '9mm'
>>> gui = MetalGUI(design)

The error comes just when I call the MetalGUI

priti-ashvin-shah-ibm, 2021-07-13:
@ani-tifr, There is a slack channel where you can ask for help.  It is in workspace  qiskit.slack.com in channel #metal.  It is an active support community for Qiskit Metal. 

Did you set up a conda environment?  The link below may be helpful to set up Metal.  It also has links to FAQ.
https://qiskit.org/documentation/metal/installation.html

marcolincs, 2021-07-13:
I believe this is documented in the FAQ: https://qiskit.org/documentation/metal/faq.html

Search for "Could not load the Qt platform" (it is Q/A 6).

Closing this case, please @ani-tifr re-open it if none of the suggestions work.

closed 2021-07-13 by marcolincs