qiskit-community/qiskit-metal #655: Add icons for all the tutorials that are missing one

zlatko-minev, 2021-07-27:
### What is the documentation modification?

Add icons for all the tutorials that are missing one

[image]

[image]
[image]

JeremyDrysdale-ibm, 2021-07-27:
None of these have graphical output.  The cards require graphical output otherwise it will use the default qiskit logo.  The exception is 4.14 which doesn't execute due to a bug that will be corrected in #648.

zlatko-minev, 2021-07-27:
Nice. We can use some of the visual from them or icons. For example, for the transmon analytics we can use some of the resulting plots from nicks notebook or just a schematic image of the transmon. 

For wavefunction for example we can show one of the images of a wavefunction etc.

PR #659 (2021-07-28): Notebook thumbnails
…phinx-thumbnail couldn't be done

### What are the issues this pull addresses (issue numbers / links)?
Closes #655 

### Did you add tests to cover your changes (yes/no)?
No

### Did you update the documentation accordingly (yes/no)?
Yes

### Did you read the CONTRIBUTING document (yes/no)?
Yes

### Summary
Utilized the direct image link approach where linking directly to a sphinx-thumbnail couldn't be done (no image in notebook to point to).

Notebook 4.14 still has the default qiskit logo thumbnail until the notebook is fixed in #648.
closed 2021-07-28 by JeremyDrysdale-ibm