qiskit-community/qiskit-metal #727: Shapely deprecation warning

parsidd, 2021-10-30:
### Information

- **Qiskit Metal version**: 0.0.4
- **Python version**: 3.9.5
- **Operating system**: Windows 10

### What is the current behavior?
Running [this](https://github.com/Qiskit/qiskit-metal/blob/main/tutorials/Appendix%20A%20Full%20design%20flow%20examples/Example%20full%20chip%20design.ipynb) tutorial notebook gives me a warning stating:
ShapelyDeprecationWarning: Setting the 'coords' to mutate a Geometry in place is deprecated, and will not be possible any more in Shapely 2.0

### Steps to reproduce the problem
Run the cells in the mentioned notebook in order. Cell 15 gives this warning the first time.

marcolincs, 2021-11-09:
Adding more history.
Shapely 1.7.1 gives no warning.
Shapely 1.8.0 starts the warning in anticipation to the future 2.0

Options:
- can limit qiskit-metal to only use shapely<1.8.0 (environment.yml and requirements.txt)
- can edit the code in qiskit-metal (and pyEPR?) where the coords are explicitly edited in place and replace it with a self-assignment or some other method.
- other option the developer can come up with.

fgnyilmaz, 2021-11-10:
Thank you @marcolincs I was going to create an issue about it as we discuss before. 
I hope it will resolve soon :)

priti-ashvin-shah-ibm, 2021-12-03:
#727

ThomasGM4, 2021-12-07:
Just to have a collection of all the warnings (in order of I think easiest to hardest to fix);

**When adding components to design**
* The 'cascaded_union()' function is deprecated. Use 'unary_union()' instead.
* Iteration over multi-part geometries is deprecated and will be removed in Shapely 2.0. Use the `geoms` property to access the constituent parts of a multi-part geometry.
* Setting the 'coords' to mutate a Geometry in place is deprecated, and will not be possible any more in Shapely 2.0
* The array interface is deprecated and will no longer work in Shapely 2.0. Convert the '.coords' to a numpy array instead.

**Ansys Render**
* No warnings so far

**GDS Render**
*

priti-ashvin-shah-ibm, 2021-12-07:
Regarding the warnings coming from Pandas:

in   https://gis.stackexchange.com/questions/397482/what-does-this-warning-mean-for-shapely-python

Historically, geopandas has used shapely to represent vector geometries. However, shapely is somewhat slow; pygeos is essentially a “vectorized” re-implementation of shapely, and can be much faster. See: https://pygeos.readthedocs.io/en/latest/#relationship-to-shapely
You might be mixing shapely and pygeos vector data. E.g. geopandas would like to store geometry via pygeos, but you might be supplying it with shapely geometries. In this case, it might convert the geometry, but due to different versions of GEOS in the background, it has to do more work to go from one or the other. That’s what the warning is telling you.

So, geopandas might be converting the shapely to pygeos, and that is what is causing some of the warnings.

zlatko-minev, 2022-01-13:
I think we now have a solution and will be uploaded here soon.

Santhosh-M31, 2022-01-31:
@zlatko-minev Waiting for the solution :)

zlatko-minev, 2022-01-31:
The team here has been working hard on this. We have solved many of the warnings, but there’s still one or two. Maybe @priti can add more

priti-ashvin-shah-ibm, 2022-01-31:
> The team here has been working hard on this. We have solved many of the warnings, but there’s still one or two. Maybe @priti can add more
> […](#)
> Sent from my iPhone

@zlatko-minev , that is correct, this is work in progress.  @Santhosh-M31 Thank you for being patient.

priti-ashvin-shah-ibm, 2023-01-31:
Merged, should be closed.

closed 2023-01-31 by priti-ashvin-shah-ibm