qiskit-community/qiskit-metal #733: Typo - the two notebooks need to be updated.

priti-ashvin-shah-ibm, 2021-11-30:
### Information

- **Qiskit Metal version**:
- **Python version**:
- **Operating system**:

Conversation in slack at #metal. 
https://qiskit.slack.com/archives/C01R8KP5WP7/p1638275287214400

### What is the current behavior?
It appears that size_x should be ‘9mm’ in the two notebooks.  The notebook in tut directory needs to be run and saved.

https://qiskit.org/documentation/metal/tut/2-From-components-to-chip/2.21-Design-a-4-qubit-full-chip.html#Let%E2%80%99s-design-the-core-of-the-chip

and
the cell has to be updated here. 

https://github.com/Qiskit/qiskit-metal/blob/main/tutorials/2%20From%20components%2[…]chip%20design/2.21%20Design%20a%204%20qubit%20full%20chip.ipynb

### Steps to reproduce the problem

### What is the expected behavior?

### Suggested solutions

Carlosbogo, 2021-11-30:
Hi, I would like to work on this. Is there anyone doing it already?

priti-ashvin-shah-ibm, 2021-11-30:
No, please feel free go ahead  and do this. 

Thanks for contributing to Metal.

priti-ashvin-shah-ibm, 2021-11-30:
https://github.com/Qiskit/qiskit-metal/blob/main/CONTRIBUTING.md

@Carlosbogo , the above links should help you.  It explains the  CLA.

Arch0125, 2021-12-06:
@priti-ashvin-shah-ibm 
Hi, I would like to work on this issue.
Is this the intended change ?
[image]

priti-ashvin-shah-ibm, 2021-12-06:
@Arch0125 , yes.  Please do follow the rest of the work needed to confirm nothing else has been affected.  

```
It appears that size_x should be ‘9mm’ in the two notebooks. The notebook in tut directory needs to be run and saved.

https://qiskit.org/documentation/metal/tut/2-From-components-to-chip/2.21-Design-a-4-qubit-full-chip.html#Let%E2%80%99s-design-the-core-of-the-chip

and
the cell has to be updated here.

https://github.com/Qiskit/qiskit-metal/blob/main/tutorials/2%20From%20components%2[…]chip%20design/2.21%20Design%20a%204%20qubit%20full%20chip.ipynb
```

Arch0125, 2021-12-06:
Yes I will start working on it.

Arch0125, 2021-12-06:
The cell link (https://github.com/Qiskit/qiskit-metal/blob/main/tutorials/2%20From%20components%2[…]chip%20design/2.21%20Design%20a%204%20qubit%20full%20chip.ipynb) is throwing server error 500 maybe because the qiskit-metal/blob directory is non-existent

priti-ashvin-shah-ibm, 2021-12-06:
> The cell link [...]

https://github.com/Qiskit/qiskit-metal/blob/main/tutorials/Appendix%20A%20Full%20design%20flow%20examples/Example%20full%20chip%20design.ipynb

This should be similar to the notebook in the doc/tut path.  The idea is that all the tutorial notebooks are in 
/tutorial directory have been run so the users can see the expected output.  Significant tutorials are in the doc/tut directories.  The idea is to confirm the tutorial notebook in /tutorial directory matches the notebook in /docs/tut.  If they are off, they need to be updated to similar.  

The notebooks in /tutorial  represent the complete set of tutorials.  The notebooks in /tutorial directory are typically not execute.

bopardikarsoham, 2022-01-11:
I've submitted a PR for this issue  @priti-ashvin-shah-ibm.

priti-ashvin-shah-ibm, 2023-01-31:
Merged, but issue not closed.

closed 2023-01-31 by priti-ashvin-shah-ibm