qiskit-community/qiskit-metal #735: Design Fails to Render in Metal GUI

SamarthH, 2021-12-02:
### Information

- **Qiskit Metal version**: 0.0.4 (pulled master branch)
- **Python version**: Python 3.9.7 | packaged by conda-forge
- **Operating system**: Tested on Windows 10 and Kubuntu 20.04

### What is the current behavior?
On trying to render a design, the metal-GUI does not draw the design, but updates the other things

Traceback (most recent call last):

  File "$HOME/testing_metal/qiskit-metal/qiskit_metal/renderers/renderer_mpl/mpl_canvas.py", line 494, in plot
    main_plot()

  File "$HOME/testing_metal/qiskit-metal/qiskit_metal/renderers/renderer_mpl/mpl_canvas.py", line 479, in main_plot
    self.clear_axis(ax)

  File "$HOME/testing_metal/qiskit-metal/qiskit_metal/renderers/renderer_mpl/mpl_canvas.py", line 541, in clear_axis
    clear_axis(ax)

  File "$HOME/testing_metal/qiskit-metal/qiskit_metal/renderers/renderer_mpl/mpl_toolbox.py", line 412, in clear_axis
    ax.lines = []

AttributeError: can't set attribute

Plotting error: can't set attribute [utility_functions.log_error_easy]

### Steps to reproduce the problem
Pull the qiskit-metal master branch
Setup metal as described in the README_developers.md using the conda method in a new environment
Try to draw any design on the gui. For example, run tutorials/Appendix A Full design flow examples/Example full chip design.ipynb
Look at metal-GUI to see the design missing

### What is the expected behavior?
The GUI should show the design

### Suggested solutions
This can be traced back to the clear_axis function in  qiskit-metal/qiskit_metal/renderers/renderer_mpl/mpl_toolbox.py
To correct this, the clear_axis method can just call axis.clear() as it did earlier

closed 2021-12-06 by SamarthH

HossamTohamy, 2022-01-20:
I tried your fix but still getting the error:

AttributeError: can't set attribute
Plotting error: can't set attribute [utility_functions.log_error_easy]

Do we comment all other commands except for: 
axis.clear()

SamarthH, 2022-01-24:
@HossamTohamy : The changed have been on the main branch. You can pull from that. Otherwise, yes, you have to comment all commands except axis.clear()