qiskit-community/qiskit-metal #747: Deprecated code in tutorial/notebook "Design 100 qubits programmatically"

koloneltedi, 2022-01-11:
## Expected Behavior
When executing block [17] of the notebook "Design 100 qubits programmatically" (link: https://qiskit.org/documentation/metal/tut/2-From-components-to-chip/2.22-Design-100-qubits-programmatically.html), one would expect the output to be as in the notebook. 

## Current Behavior
The following error-message: 
"This method is deprecated. Change your scripts to useanalysis.quantitative.capacitance_lom.run_lom(). 
This is logical since this is exactly what the source-code of the method currently says."

## System Settings
Qiskit Metal        0.0.4

Basic
____________________________________
 Python              3.7.8 | packaged by conda-forge | (default, Nov 27 2020, 18:48:03) [MSC v.1916 64 bit (AMD64)]
 Platform            Windows AMD64
 Installation path   c:\users\sasha\documents\discoveries\qiskitmetal\source\qiskit_metal

Packages
____________________________________
 Numpy               1.19.5
 Qutip               4.6.0

Rendering
____________________________________
 Matplotlib          3.4.1

GUI
____________________________________
 PySide2 version     5.13.2
 Qt version          5.9.7
 SIP version         4.19.8

IBM Quantum Team

## Steps to Reproduce
Follow the notebook. 

## Detailed Description

Presumably one could rewrite the tutorial with the method descibed in the error-message (useanalysis.quantitative.capacitance_lom.run_lom()). However I haven't found any documentation or source for that method, so am not sure if that actually exists...
When looking ahead to future tutorials, I saw that block [14] of tutorial "Capacitance matrix and LOM analysis" (link: https://qiskit.org/documentation/metal/tut/4-Analysis/4.01-Capacitance-and-LOM.html) gave similar output to what was expected. Now I haven't run that tutorial yet (I'm just beginning with Metal), so am not sure if that is exactly the same or whether it works, but that seems to solve it. 

## Possible Implementation

zlatko-minev, 2022-01-13:
Thanks for dropping the issue. Any chance you want to make a pull request with the fix?

koloneltedi, 2022-01-13:
I'd love to! Will not be in the coming few days though, since I realized that my current Ansys license doesn't seem to support q3d, so I can't do the necessary testing. But once I fixed that I will look into it. Just plan to follow tutorial 4.1, LOM part. So if you don't mind the time-delay, I'll be looking forward to working on the fix. Seems like a good practise for contributing some new qubit components and such.

zlatko-minev, 2022-01-17:
Great to hear, yes, no worries on the time. Looking forward to your pull request

koloneltedi, 2022-01-25:
I finally got to it, hopefully it is a stepping stone for any 'real' code contributions to come haha

priti-ashvin-shah-ibm, 2023-01-31:
This has merged.

closed 2023-01-31 by priti-ashvin-shah-ibm