qiskit-community/qiskit-metal #798: Updating qiskit-metal to use PySide6

shanto268, 2022-05-30:
Please be as descriptive as possible, including: what is expected, why is this feature needed, what is the objective, etc.

Installing qiskit-metal on M1 Macs is a nightmare for many reasons as y'all are mostly well aware of. However, y'all can help the process by simply updating the code base to be using PySide6 as opposed to PySide2. This small simple change will make qiskit-metal run natively on Apple Silicon.

### What is the feature being requested?

Migrating codebase from **PySide2** to **PySide6**.

### What are use cases for this feature?

Makes qiskit-metal run natively on newer Macs (i.e. M1 Apple Silicon)

priti-ashvin-shah-ibm, 2022-08-02:
@shanto268 , We are happy to have more people using Qiskit Metal.  

Did you follow option 1A:
https://qiskit.org/documentation/metal/installation.html

This usually helps also:
https://qiskit.org/documentation/metal/faq.html

This may help:
https://qiskit.org/documentation/metal/

Using Slack, in the Qiskit workspace, have you joined #metal?    That is where our community asks a variety of questions.  You can also search older questions to get answers for previously asked questions.  

We are aware of pyside6.  Metal is open source code and we welcome the community to contribute to the repository. 

Presently, to work on EPR using Ansys , one needs to use a commercial license for Ansys.  The current renderer is using Windows comm ports.  For EPR we are working on adding an option to use pyaedt vs comm port, along with using free simulation software which does not require Ansys.  There are discussions (#metal) and recorded tutorials on YouTube channel (https://github.com/Qiskit/qiskit-metal/blob/main/README_Tutorials.md) about the free simulation software that you can reference.

shanto268, 2022-08-02:
Hi @priti-ashvin-shah-ibm thanks for your response. I did follow all the resources you mentioned but they do not really tackle the issue that I raised....

I am yet to join that channel on Slack but I will surely do that. Thanks for the recommendation and also for the clarification of the issue.

For the time being I will point to this gist that I created to get around the issue in the event that it may help others - [https://gist.github.com/shanto268/99ea3ce60d61e10a00e292f548b59c39 ](https://gist.github.com/shanto268/99ea3ce60d61e10a00e292f548b59c39 )

I will look into patching the issue once I have some of that valuable free time on my hand 😊

obrienpja, 2023-02-07:
I am addressing this issue in this pull request: #908