qiskit-community/qiskit-metal #821: Website Broken

Yuu6883, 2022-07-25:
### Information

- **Qiskit Metal version**:
- **Python version**:
- **Operating system**:

### What is the current behavior?
The website is broken: https://qiskit.org/documentation/metal/circuit-examples/index.html# (or any link in the same directory)

### Steps to reproduce the problem
https://qiskit.org/documentation/metal/circuit-examples/index.html# 

### What is the expected behavior?
Site working

### Suggested solutions
[commit fb9a39](https://github.com/Qiskit/qiskit-metal/commit/fb9a391f20ad83637dc769382bc8809623475ad9) probably broke the site generator

priti-ashvin-shah-ibm, 2022-07-25:
Thanks for the heads up.  We will work on it.  As a workaround you can get access to ALL tutorial notebooks at:
https://github.com/Qiskit/qiskit-metal/tree/main/tutorials

The difference between these document and what is in docs is that within docs, the notebooks show the output.

priti-ashvin-shah-ibm, 2022-07-25:
This directory has the executed notebooks are at:

https://github.com/Qiskit/qiskit-metal/tree/main/docs/tut

closed 2022-07-26 by priti-ashvin-shah-ibm

Yuu6883, 2022-07-26:
Thanks for the fix, it works fine now.