qiskit-community/qiskit-metal #823: Broken Jupyter notebook links in Overview Section of Tutorials page

bopardikarsoham, 2022-07-27:
### What is the current behavior?
The three modules in the overview section of the [Tutorials](https://qiskit.org/documentation/metal/tut/index.html) page:

1) [Bird’s eye view of Qiskit Metal](https://qiskit.org/documentation/metal/tut/1-Overview/1.1-Bird%27s-eye-view-of-Qiskit-Metal.html)
2) [Quick start](https://qiskit.org/documentation/metal/tut/1-Overview/1.2-Quick-start.html)
3) [Save your chip design](https://qiskit.org/documentation/metal/tut/1-Overview/1.3-Saving-Your-Chip-Design.html)

Each has a jupyter notebook link in the **NOTE**  section which is broken

### Suggested solutions

The links should redirect to the Qiskit Metal GitHub tutorials page:

1) https://github.com/Qiskit/qiskit-metal/blob/main/docs/tut/1-Overview/1.1-Bird's-eye-view-of-Qiskit-Metal.ipynb

2) https://github.com/Qiskit/qiskit-metal/blob/main/docs/tut/1-Overview/1.2-Quick-start.ipynb

3) https://github.com/Qiskit/qiskit-metal/blob/main/docs/tut/1-Overview/1.3-Saving-Your-Chip-Design.ipynb

**OR**

There is actually a single hyphen missing in all of these links. Adding this will fix the issue.

[image]

priti-ashvin-shah-ibm, 2022-07-27:
Hi @bopardikarsoham , thank you for helping our docs look at their best.  There was a recent release of sphinx for docs. We consequently have updated few commits to address the changes.  Could you get the latest commit on main and see if you still have this issue? For me, these links work and I am not able to repeat this issue.

bopardikarsoham, 2022-07-28:
I went on the qiskit.org/metal site after that I clicked on the tutorials present in the 'Overview' Section and I saw that it's not working on the frontend

bopardikarsoham, 2022-07-28:
Still it does not work with the recent release of sphinx I guess at least for me

priti-ashvin-shah-ibm, 2022-07-28:
I saw it with @obrienpja , we are pusing a fix.

priti-ashvin-shah-ibm, 2022-07-28:
@bopardikarsoham , do you still see the problem?

closed 2022-07-28 by obrienpja