qiskit-community/qiskit-metal #870: Request: Add styles for the TransmonCrossFL, allows us to accommodate different types of flux lines.

clarkmiyamoto, 2022-11-02:
Please be as descriptive as possible, including what is expected, why is this feature needed, what is the objective, etc.

### What is the feature being requested?

1. In the `default_options` of the `TransmonCrossFL`, add a type of flux line style.
2. Add a style for MIT's Lincon Labs which allows us to accommodate

### What are use cases for this feature?
A lot of researchers send their designs to MIT's Lincon Labs for fabrication. Having this geometry baked into Qiskit Metal should speed up the design + fabrication process!

clarkmiyamoto, 2022-11-02:
I'll be working on this, will push in a few days.

priti-ashvin-shah-ibm, 2022-11-02:
> I'll be working on this, will push in a few days.

Glad to see active contributors to Metal.