qiskit-community/qiskit-metal #920: New LOM: Transmon Transition Energy

schwarz11, 2023-02-10:
### Information

- **Qiskit Metal version**: 0.1.2
- **Python version**: 3.11.0 | packaged by conda-forge | (main, Oct 25 2022, 06:12:32) [MSC v.1929 64 bit (AMD64)]
- **Operating system**: Windows-10-10.0.19044-SP0

### What is the current behavior?
[image]
Too large energy levels (order of 10^3 GHz)

### Steps to reproduce the problem
Inspired from the `Qiskit_Metal\qiskit-metal\tutorials\4 Analysis\A. Core - EM and quantization\4.04 New LOM and Fluxonium Example.ipynb`, sweep a transmon qubit with $n_g$. The transmon parameters are based on `Transmon_5p5GHz_fQ_cmat.txt`
```
HilbertSpace:  subsystems
-------------------------

Transmon------------| [Transmon_2]
                    | EJ: 13278.758148398147
                    | EC: 335.4436048599739
                    | ng: 0.001
                    | ncut: 150
                    | truncated_dim: 10
                    |
                    | dim: 301
```
### What is the expected behavior?
[image]

priti-ashvin-shah-ibm, 2023-02-10:
Hi @schwarz11 , can you add more information?

Please show the code you used to sweep.  Also, did you recently make a conda env?  Was it October 25th?

If you don't get the results you want, it could be the components within your design, vs the software repo.  If you want help to use Metal, discussions are in Slack. You need to join Qiskit workspace, then ask on #metal.

schwarz11, 2023-02-12:
Hi @priti-ashvin-shah-ibm , thank you for the reply!

Here is the code I used for the results above. Basically, I am trying to sweep charge to transmon, in the same way as sweeping flux to fluxonium in the tutorial. I have also checked that my scqubits is up-to-date (version 3.1.0) in the environment.

[image]
[image]
[image]
[image]
[image]

zachparrott, 2023-03-30:
It appears the EJ and EC values in your scqubits Transmon object are in MHz*h not GHz*h, this is giving the scaling of 1e3. 

```
Python 3.9.16 (main, Mar  8 2023, 10:39:24) [MSC v.1916 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import scqubits as scq
>>> tmon1 = scq.Transmon(13278, 335, 0.001, 150)
>>> ev1 = tmon1.eigenvals()
>>> ev1 - ev1[0]
array([    0.        ,  5608.01726784, 10821.28811915, 15597.19000465,
       19612.15215931, 23961.14733016])
>>> # now rescale
>>> tmon2 = scq.Transmon(13.278, 0.335, 0.001, 150)
>>> ev2 = tmon2.eigenvals()
>>> ev2 - ev2[0]
array([ 0.        ,  5.60801727, 10.82128812, 15.59719   , 19.61215216,
       23.96114733])
```

schwarz11, 2023-04-02:
Hi @zachparrott , it seems that you're right. Then, I think special care needs to be performed for units when transferring from transmon to composite system before calculating the `hilbertspace` , hence `qubit`.

zlatko-minev, 2026-09-27:
Following up on this older thread: @zachparrott's diagnosis is correct and still applies on current `main`. In the new LOM, the `TRANSMON` subsystem builds `scqubits.Transmon` with EJ and EC in **MHz** (`Ej_from_Lj` / `Ec_from_Cs` default to MHz), so eigenvalues and transitions come out in MHz even though scqubits' unit label defaults to GHz. With the 4.04 `Transmon_5p5GHz_fQ_cmat.txt` data I get EJ ≈ 13268, EC ≈ 335.4 and E1−E0 ≈ 5609, i.e. f01 ≈ 5.61 GHz. Calling `scq.set_units("MHz")` before plotting, or dividing by 1000, gives the expected picture.

We'll also correct the `Subsystem` docstring, which currently says the energies are in GHz. Closing as resolved — feel free to reopen if you still see a discrepancy after accounting for units.

---
_Generated by [Claude Code](https://claude.ai/code)_

closed 2026-09-27 by zlatko-minev