qiskit-community/qiskit-metal #922: cpw_gap bug in rendering qiskit-metal design to GDS file.

patrick612, 2023-02-28:
### Information

- **Qiskit Metal version**: 0.0.4
- **Python version**: 3.9.12
- **Operating system**: Windows 11

### What is the current behavior?
The cpw_gap does not render correctly when exported to GDS. Specifically, although the fillet of the trace is there, cpw_gap has no fillet and has square edges. Furthermore, the width of cpw_gap seems to have doubled when rendered on GDS.

### Steps to reproduce the problem
Make any circuits with cpw or any elements with etched gap, then export the qiskit-metal design to a GDS file.

### What is the expected behavior?
It is expected to see the fillet and correct width on etched gap of design elements like cpw.

### Suggested solutions

priti-ashvin-shah-ibm, 2023-03-02:
If that doesn't work, can you copy/paste the code you used?   To duplicate your situation. I don't want to re-type your code.

patrick612, 2023-03-02:
Changing the view_in_file to False in the no_cheese options worked!! Thank you!

priti-ashvin-shah-ibm, 2023-03-02:
Super.  I will close this issue.

closed 2023-03-02 by priti-ashvin-shah-ibm