qiskit-community/qiskit-metal #933: ElmerFEM error: " FileNotFoundError: [WinError 3] The system cannot find the path specified: 'out' "

sirshisram, 2023-04-07:
### Information

- **Qiskit Metal version**: 0.1.2
- **Python version**: 3.11.2
- **Operating system**: Windows 10 Pro 22H2 (OS Build: 19045.2788)

### What is the current behavior?

[image]

Was going through the analysis tutorial notebook : "4.19 Analyze a transmon using ElmerFEM.ipynb" 
Gmsh works properly when called. Installed ElmerFEM and verified it on the terminal.

### Steps to reproduce the problem

### What is the expected behavior?

### Suggested solutions

diemilio, 2023-04-08:
Hi @sirshisram. This is the kind of error you would get if there is not mesh file for ElmerFEM to run. In the directory where the notebook is located, do you see the file `out.msh`? If not, what happens when you run the cell with `elmer_renderer.export_mesh()`?

I believe this is a mesh export issue, but just FYI, I also did a fresh install on a Windows PC with no dependencies previously installed (no anaconda, no qiskit-metal, no gmsh, no ElmerFEM, etc.), and was able to successfully run the notebook. Here are the steps I followed:

1. Installed Anaconda
2. Installed Microsoft C++ Build Tools
3. Installed ElmerFEM (select add path for current user during installation) 
4. In anaconda prompt `conda install -c anaconda git`
5. `cd` into folder to install metal
6. `git clone https://github.com/Qiskit/qiskit-metal.git`
7. `cd qiskit-metal`
8. `conda create --name qiskit-metal-src --file=environment.yml`
9. `conda activate qiskit-metal-src`
10. `python -m pip install --no-deps -e .`
11. `conda install ipykernel`
12. `ipython kernel install --user --name=qiskit-metal-src`
13. Opened gmsh/elmerfem tutorial notebook and ran all cells (not issues encountered).

lmk if you have any questions.

sirshisram, 2023-04-08:
Hi @diemilio! Thanks for your response!

I did check the directory of the notebook and out.msh is there. For the safer side, I deleted it and rerun the whole notebook again.

Upon executing **elmer_renderer.export_mesh()**, it does create **out.msh**. No error comes out.

[image]

[image]

Yet, it outputs the same error: **FileNotFoundError: [WinError 3] The system cannot find the path specified: 'out'**

diemilio, 2023-04-08:
Hi @sirshisram, in the simdata folder, are there `elmergrid.txt` and `elmersolver.txt` files?

I might need to take a look at those. Are you subscribed to the Qiskit slack Workspace? Could you please share this issue in the #metal channel in there? and upload those files. Thanks.

sirshisram, 2023-04-08:
Sure @diemilio.

diemilio, 2023-04-11:
After a series of discussions in qiskit's #metal slack channel, we have identified that this issue is not related to any of the code within the `qiskit-metal` repo; it seems to have to do with `ElmerGrid`, which is part of an external tool (ElmerFEM).

For this reason I think we can close this github issue and continue the discussion in the slack workspace.

sirshisram, 2023-04-11:
sure @diemilio

closed 2023-04-11 by sirshisram

sirshisram, 2023-04-11:
I completely uninstalled ElmerFEM, and was able to make it work by installing `ElmerFEM-gui-mpi-Windows-AMD64`, instead of installing `ElmerFEM-gui-nompi-Windows-AMD64`.