qiskit-community/qiskit-metal #945: Dependencies pinned too tightly

wshanks, 2023-05-22:
#921 pinned all of the versions in requirements.txt to specific versions in order to avoid test failures. It makes sense to pin the dependency versions in tests to ensure consistency, but requirements.txt also gets used to pass to `install_requires` of setuptools to set the dependency versions of the package. Having all the dependencies pinned to specific versions (rather than looser ranges) makes it difficult to use qiskit-metal in an environment with other packages because all of those other packages have to be compatible with all of the pinned versions of qiskit-metal's dependencies.

If you want the tests to run against fixed dependency versions, one option is to add another requirements file named something like requirements-frozen.txt or requirements-pinned.txt, with all of the dependencies pinned while leaving looser ranges in requirements.txt. Then in tox.ini, you can put `-r requirements-pinned.txt` under the `deps` entry. Something to keep in with this approach is that if you allow newer versions of dependencies in requirements.txt but are not testing against them in CI you will not discover issues with new versions of the dependencies until a user complains.

zlatko-minev, 2023-06-14:
Thank you, Will. We should do this. 

I'm looking for help from folks here to try this?

mulliganaceous, 2024-11-15:
I might also be able to help. Right now it also seems I could not get Qiskit Metal working unless I am using Python 3.10. It required me to modify the YML file itself by replacing '>' with '=='.

zlatko-minev, 2024-11-17:
So you set it to 3.10

mulliganaceous, 2024-11-18:
> So you set it to 3.10

Yep. I set it to 3.10. I could not use Metal in 3.11 because of unresolvable incompatibility, especially regarding Sphin.x