qiskit-community/qiskit-metal #966: Bug documentation website render

soosub, 2023-07-06:
### What is the documentation modification?
There seems to be an issue when deploying the documentation website for Qiskit Metal. See screenshots below or e.g. [the example on resonator meandering](https://qiskit.org/documentation/metal/circuit-examples/B.Resonators/11-Resonator_Meander.html). As a quick fix, checking out the notebooks directly on github is fine but it would be nice to view them normally on the website too.

Cheers!

[image]
[image]

q-inho, 2023-11-29:
I think this problem also has been solved after this [PR](https://github.com/qiskit-community/qiskit-metal/pull/975) merged.

closed 2023-11-29 by zlatko-minev

soosub, 2023-12-11:
Great, many thanks @q-inho !