qiskit-community/qiskit-metal #972: Error trying to install qiskit_metal

Marcofusco9, 2023-09-25:
### Information

Hi everyone, i'm trying to install qiskit_metal in windows. i've followed the step in the documentation : creating a repository on GIT and cloning it in a folder on my pc. Then i navigate in the folder 
[image]
cd-> folder but when i write "conda env create -n <env_name> environment.yml" gives an error : file not found. Of course the file enivrronment.yml is present in the related folder. Any advice ? thanks. 

### What is the current behavior?

### Steps to reproduce the problem

### What is the expected behavior?

### Suggested solutions

q-inho, 2023-11-26:
Have you try to replace the <env_name> into your environment name you want to make?
For example try `conda env create -n METAL environment.yml` to name your environment "METAL".

zlatko-minev, 2023-11-27:
Ditto on @q-inho 's suggestion

abhishekchak52, 2025-12-14:
Closing as potentially resolved.

closed 2025-12-14 by abhishekchak52