qiskit-community/qiskit-metal #982: Help on adding datatypes as an option to qgeometries

maxhant, 2024-01-16:
I want to have the option to choose which GDS datatype each geometry will belong to. For that, I have tried editing the render as follows:

```python
    def _qgeometry_to_gds(
        self, qgeometry_element: pd.Series
    ) -> Union['gdspy.polygon', 'gdspy.FlexPath', None]:
        """
         Docs
        """

        # pylint: disable=too-many-locals

        corners = self.options.corners
        tolerance = self.parse_value(self.options.tolerance)
        precision = self.parse_value(self.options.precision)
        max_points = int(self.parse_value(self.options.max_points))

        geom = qgeometry_element.geometry  # type: shapely.geometry.base.BaseGeometry

        if isinstance(geom, shapely.geometry.Polygon):
            exterior_poly = gdspy.Polygon(
                list(geom.exterior.coords),
                layer=qgeometry_element.layer,
                datatype=qgeometry_element.sublayer,
            )
```

However, it does not work and I don't really understand why. It has an impact which is that all geometries are now on datatype 0. Can someone point me to the right direction? 

### What is the feature being requested?
Being able to expose the same layer with different intensities. Having them on separate layers does not work, else the cheesing will ignore the feature on higher layers. I have been discussing this on [Slack](https://qiskit.slack.com/archives/C01R8KP5WP7/p1705243351105309). 

### What are use cases for this feature?
Having a more advanced fab.

maxhant, 2024-01-17:
Hi @zlatko-minev, thanks for your quick answer. I have created a draft PR #983 for you to see what I have done. Please, keep in mind that I am a novice at all this, but I am happy to learn to make it right!

zlatko-minev, 2024-01-17:
Great job, very well done! This is a substantial change part of the code so we should take this a little carefully. 

Have you checked this in one or two settings to make sure it works and doesn't fail anywhere? 

Is it backwards compatible

maxhant, 2024-01-18:
> Have you checked this in one or two settings to make sure it works and doesn't fail anywhere?

For now, it does not work, so it is already failing. I was hoping that you would know what am I missing to have it working. 

Once it works, which settings do you want me to test and I will test it properly?

maxhant, 2024-01-18:
I suspect that there is something to change in the `qiskit-metal\qiskit_metal\toolbox_metal\layer_stack_handler.py`