qiskit-community/qiskit-metal #984: Set up GitHub Pages

arnaucasau, 2024-02-05:
### Instructions 
The project now has the documentation deployed in a branch called [gh-pages](https://github.com/qiskit-community/qiskit-metal/tree/gh-pages), but we need to set up the repo to use GitHub Pages by following the next 5 steps:

1. Navigate to your site's repository
2. Click `Settings`
3. Go to `Pages`
4. Select `Deploy from a branch`, under the `Source` section on `Build and deployment`
5. Select the `gh-pages` branch and the `/ (root)` folder in the `Branch` section on `Build and deployment`

[image]

The default URL where you can find your site is https://qiskit-community.github.io/qiskit-metal

For further information, you can see the [GitHub Pages Quickstart](https://docs.github.com/en/pages/quickstart#creating-your-website).

zlatko-minev, 2024-02-07:
Thank you! Ok, so i just did this, what's next 
[image]

zlatko-minev, 2024-02-07:
We will want to merge this branch in the main sometime?

arnaucasau, 2024-02-07:
Thank you @zlatko-minev! The page is now working at https://qiskit-community.github.io/qiskit-metal :tada: 

No need to merge the branch, the `gh-pages` branch contains the docs built, and thanks to the new workflow it will be updated at each push to the main, similar to how we deployed to qiskit.org before :)

The only thing left in the issue is to change the link in the About section of the repo to show the new one.

[image]

P.S. If someday you want to deploy the docs from a stable version, the deployment changes introduced in this [commit](https://github.com/qiskit-community/qiskit-metal/commit/5fd9379e22dacdf4b7bf7d67eb327b969c31ed13) will need to be backported

zlatko-minev, 2024-02-07:
Awesome, thank you! Updated!

closed 2024-02-07 by arnaucasau