qiskit-community/qiskit-metal #987: Error in effective EC of an individual transmon when two identical transmons are meshed together as one cell in LOM analysis.

sirshisram, 2024-03-23:
### Information

- **Qiskit Metal version**: 0.1.5
- **Python version**: 3.11.4
- **Operating system**: Windows 10 (x64)

### What is the current behavior?

The effective EC that I obtain when I mesh for individual transmons in my design is ~456 MHz. 

[image]

When I mesh two such identical transmons (who are individually having EC = 456 MHz) together as one cell, I get EC to be ~227 MHz, no matter how far they are separated. 

[image]

[image]

I varied the distance between the transmons from 1mm to 5mm and the total EC barely changed.  This does not make sense to me as changing the island pad gaps significantly changed the EC for the transmon (when meshed individually). 

The capacitances that I get from meshing one transmon is identical to what I get from meshing two transmons together, so it does not seem to be an issue from ElmerFEM.

[image]

[image]

### Steps to reproduce the problem

Mesh two identical transmons together as one cell.

### What is the expected behavior?

The effective EC of a transmon when meshed individually should be comparable to the effective EC when meshing two transmons together.

Please help me in this regard. Thanks!

zlatko-minev, 2024-03-23:
Sorry, what do you mean by mesh the two tmons as one cell. Each of them should be a separate Cell. Or if you make them as one object, then they would be one system.

sirshisram, 2024-03-24:
My terminology might be wrong, I mean meshing the two tmons together and obtaining the cross-capacitances. I define them as two separate cells.

sirshisram, 2024-05-15:
@zlatko-minev would it be possible for you to look into this issue? thanks in advance.

zlatko-minev, 2024-05-15:
A bit caught up at the moment. What do you mean by mashing them together separately? How are you doing this? Some code or the like would help

sirshisram, 2024-05-16:
I have a quantum circuit that consists of two tmons. As per the tutorial video by Yehan, you had mentioned that one can mesh and obtain the capacitance matrix for the whole quantum circuit at one shot, but that would be quite the load computationally. doing so would yield cross capacitances among the metal pads of the two qubits. Im running into an issue where Im getting a 
result different from what I got when one would individually (one tmon at a time) obtain the capacitance matrix. and if i also vary the distance between the two tmons, there seems to be no change in the EC of the tmon whatsoever. my knowledge says that shouldn't happen. i hope my problem is clearer now.

i define the two tmons and mesh them together:

`elmer_renderer = QElmerRenderer(design)`
`elmer_renderer.gmsh.options.mesh.min_size = "5um" `
`elmer_renderer.gmsh.options.mesh.max_size = "50um"`
`elmer_renderer.render_design(mesh_geoms=True, open_pins=[("Q1", "coup1"), ("Q1", "read1"), ("Q2", "coup2"), ("Q2", "read2")], skip_junctions=True)`
`elmer_renderer.export_mesh()`
`elmer_renderer.add_solution_setup('capacitance')`
`elmer_renderer.run('capacitance')`
`capmax = elmer_renderer.capacitance_matrix`

i obtain the total capacitance matrix of the ckt:

[image]

now i define the cells:

`opt1 = dict(node_rename = {'Q1_read1_connector_pad': 'read1', 'Q1_coup1_connector_pad': 'coupling'}, `
`            cap_mat = capmax, ind_dict = {('Q1_pad_top', 'Q1_pad_bot'):28}, jj_dict = {('Q1_pad_top', 'Q1_pad_bot'):'j1'})`
`cell_1 = Cell(opt1)`

`opt2 = dict(node_rename = {'Q2_read2_connector_pad': 'read2', 'Q2_coup2_connector_pad': 'coupling'},`
`            cap_mat = capmax, ind_dict = {('Q2_pad_top', 'Q2_pad_bot'):28}, jj_dict = {('Q2_pad_top', 'Q2_pad_bot'):'j2'})`
`cell_2 = Cell(opt2)`

i get a EC of 227MHz or so.

but if i individually mesh each tmon

`elmer_renderer = QElmerRenderer(design)`
`elmer_renderer.gmsh.options.mesh.min_size = "5um"; elmer_renderer.gmsh.options.mesh.max_size = "50um";`
`elmer_renderer.render_design(mesh_geoms=True, selection=['Q1'], open_pins=[("Q1", "coup1"), ("Q1", "read1")], skip_junctions=True)`
`elmer_renderer.export_mesh()`
`elmer_renderer.add_solution_setup('capacitance')`
`elmer_renderer.run('capacitance')`
`t1_matrix = elmer_renderer.capacitance_matrix`

`elmer_renderer.render_design(mesh_geoms=True, selection=['Q2'], open_pins=[("Q2", "coup2"), ("Q2", "read2")], skip_junctions=True)`
`elmer_renderer.export_mesh()`
`elmer_renderer.add_solution_setup('capacitance')`
`elmer_renderer.run('capacitance')`
`t2_matrix = elmer_renderer.capacitance_matrix`

and defining the cells and subsystems then, i get EC of 455MHz or so.

what my problem is that is this difference shouldn't be this large (we expect some capacitive dressing, but this big difference seems to be an error). and what confirms that this is an error, at least for me, is that if i vary the distance between the tmons, the EC does not change whatsoever. i say this is an error because i could study a significant change in EC when i played around with the metal pads inside the tmon circuit.

please correct me if i have gone wrong anywhere. thank you in advance.

sirshisram, 2024-06-01:
@zlatko-minev i know you're really busy, but is it possible for you to look into this issue?  i have responded with what i find the issue i am running into LOM. thanks in advance.