qiskit-community/qiskit-metal #988: AttributeError: module 'numpy' has no attribute 'float'.

samuelkim16, 2024-05-09:
### Information

- **Qiskit Metal version**: 0.1.5
- **Python version**: 3.11.9
- **Operating system**: Windows

### What is the current behavior?

Running the [tutorial here](https://github.com/qiskit-community/qiskit-metal/blob/main/tutorials/4%20Analysis/A.%20Core%20-%20EM%20and%20quantization/4.02%20Eigenmode%20and%20EPR.ipynb) gives the below error in the last cell. The requirements list `numpy==1.24.2` so `np.float` should be avoided. 

```python
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Cell In[52], line 1
----> 1 eig_2qb.run_epr()

File c:\users\username2\documents\qiskit-metal\qiskit_metal\analyses\quantization\energy_participation_ratio.py:173, in EPRanalysis.run_epr(self, no_junctions)
    171 if not no_junctions:
    172     self.run_analysis()
--> 173     self.spectrum_analysis(self.setup.cos_trunc, self.setup.fock_trunc)
    174     try:
    175         self.report_hamiltonian(self.setup.sweep_variable)

File c:\users\username2\documents\qiskit-metal\qiskit_metal\analyses\quantization\energy_participation_ratio.py:225, in EPRanalysis.spectrum_analysis(self, cos_trunc, fock_trunc)
    221 def spectrum_analysis(self, cos_trunc: int = 8, fock_trunc: int = 7):
    222     """Short-cut to the same-name method found in renderers.ansys_renderer.py.
    223     Eventually, the analysis code needs to be only here, and the renderer method deprecated.
    224     """
--> 225     self.sim.renderer.epr_spectrum_analysis(cos_trunc, fock_trunc)

File c:\users\username2\documents\qiskit-metal\qiskit_metal\renderers\renderer_ansys\ansys_renderer.py:1770, in QAnsysRenderer.epr_spectrum_analysis(self, cos_trunc, fock_trunc)
   1762 """Core epr analysis method.
   1763 
   1764 Args:
   1765     cos_trunc (int, optional): truncation of the cosine. Defaults to 8.
   1766     fock_trunc (int, optional): truncation of the fock. Defaults to 7.
   1767 """
   1768 self.epr_quantum_analysis = epr.QuantumAnalysis(
   1769     self.epr_distributed_analysis.data_filename)
-> 1770 self.epr_quantum_analysis.analyze_all_variations(cos_trunc=cos_trunc,
   1771                                                  fock_trunc=fock_trunc)

File ~\AppData\Local\anaconda3\envs\qiskit\Lib\site-packages\pyEPR\core_quantum_analysis.py:461, in QuantumAnalysis.analyze_all_variations(self, variations, analyze_previous, **kwargs)
    459         result[variation] = self.results[variation]
    460     else:
--> 461         result[variation] = self.analyze_variation(variation, **kwargs)
    464 self.results.save()
    466 return result

File ~\AppData\Local\anaconda3\envs\qiskit\Lib\site-packages\pyEPR\core_quantum_analysis.py:682, in QuantumAnalysis.analyze_variation(self, variation, cos_trunc, fock_trunc, print_result, junctions, modes)
    680 # Numerical diag
    681 if cos_trunc is not None:
--> 682     f1_ND, CHI_ND = epr_numerical_diagonalization(freqs_hfss,
    683                                                   Ljs,
    684                                                   PHI_zpf,
    685                                                   cos_trunc=cos_trunc,
    686                                                   fock_trunc=fock_trunc)
    687 else:
    688     f1_ND, CHI_ND = None, None

File ~\AppData\Local\anaconda3\envs\qiskit\Lib\site-packages\pyEPR\calcs\back_box_numeric.py:66, in epr_numerical_diagonalization(freqs, Ljs, φzpf, cos_trunc, fock_trunc, use_1st_order, return_H, non_linear_potential)
     61 assert(all(freqs < 1E6)
     62        ), "Please input the frequencies in GHz. \N{nauseated face}"
     63 assert(all(Ljs < 1E-3)
     64        ), "Please input the inductances in Henries. \N{nauseated face}"
---> 66 Hs = black_box_hamiltonian(freqs * 1E9, Ljs.astype(np.float), fluxQ*ϕzpf,
     67                            cos_trunc, fock_trunc, individual=use_1st_order,
     68                            non_linear_potential = non_linear_potential)
     69 f_ND, χ_ND, _, _ = make_dispersive(
     70     Hs, fock_trunc, ϕzpf, freqs, use_1st_order=use_1st_order)
     71 χ_ND = -1*χ_ND * 1E-6  # convert to MHz, and flip sign so that down shift is positive

File ~\AppData\Local\anaconda3\envs\qiskit\Lib\site-packages\numpy\__init__.py:305, in __getattr__(attr)
    300     warnings.warn(
    301         f"In the future `np.{attr}` will be defined as the "
    302         "corresponding NumPy scalar.", FutureWarning, stacklevel=2)
    304 if attr in __former_attrs__:
--> 305     raise AttributeError(__former_attrs__[attr])
    307 # Importing Tester requires importing all of UnitTest which is not a
    308 # cheap import Since it is mainly used in test suits, we lazy import it
    309 # here to save on the order of 10 ms of import time for most users
    310 #
    311 # The previous way Tester was imported also had a side effect of adding
    312 # the full `numpy.testing` namespace
    313 if attr == 'testing':

AttributeError: module 'numpy' has no attribute 'float'.
`np.float` was a deprecated alias for the builtin `float`. To avoid this error in existing code, use `float` by itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use `np.float64` here.
The aliases was originally deprecated in NumPy 1.20; for more details and guidance see the original release note at:
    https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations
```

### Suggested solutions

zlatko-minev, 2024-05-10:
Thank you! The solution seems to use float not np float Want to make a pull request? Much appreciated

    https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations

vanitery, 2024-05-14:
It's a common bug. I've made a few pull requests here and there replacing np.float with float (python has it natively), and we just need to weed out the rest. Stuff like this happens when things update.

Go for a pull request where you change that little detail in the relevant bit of source code and you should be golden.

zlatko-minev, 2024-06-07:
> I misread the error message. It is actually an issue in pyEPR, which has been fixed as far as I can tell in the latest version (0.9.0). The requirements for qiskit-metal list pyEPR-quantum==0.8.5.7. Would it be possible to update this?

Would Yes, but would you mind testing it locally first before we make a global change?

To test using qiskit-metal with the new pyEPR version, you can follow these steps.
(If you want, you can use conda or virtual environment isolate your environment and manage dependencies effectively.)

After installing metal, install the desired version of pyEPR:
pip install pyEPR==desired_version

If needed, you can try this too to resolve version conflcits, and to force install it while ignoring its pyEPR dependency.
pip install pyEPR==desired_version --no-deps

samuelkim16, 2024-06-10:
Yes, I have tested it in the above tutorial, and it runs without error.

zlatko-minev, 2025-12-07:
Solved by v0.5