qiskit-community/qiskit-metal #990: gdspy package not available through the conda-forge channel

arantxapardue, 2024-05-21:
### Information

- **Qiskit Metal version**: 0.1.5 (2023)
- **Python version**: 3.11.9
- **Operating system**: macOS Sonoma 14.4.1

### What is the current behavior?
When creating a new environment from the terminal in VS Code, the environment.yml file cannot be implemented all of the way due to gdspy (version 1.6.12) not being available through the conda-forge channel.

### Steps to reproduce the problem
From base environment:
`conda env create -n env4-dev --file environment.yml`
Which gives error:
`PackagesNotFoundError: The following packages are not available from current channels:`
 ` - gdspy==1.6.12`

### What is the expected behavior?
Should create an environement with all packages listed in environment.yml file.

### Suggested solutions
Maybe use a different channel.

arantxapardue, 2024-06-17:
fixed with
`conda install -c conda-forge gdspy`

closed 2024-06-17 by arantxapardue

zlatko-minev, 2024-06-23:
Great, thank you for the update

MGKincart, 2024-11-03:
I am having the same issue but using conda install -c conda-forge gdspy didn't fix it

zlatko-minev, 2024-11-07:
Not sure what you mean. You have to give more info.

Try with pip: pip install gdspy==1.6.12

MGKincart, 2024-11-07:
I am installing on mac and running into this issue while trying to create
the environment. I have tried that and it returned that that version no
longer exists.

MGKincart, 2024-11-08:
> Did you try just pip?
> Or here https://anaconda.org/conda-forge/gdspy

Yea I can install it but the environment needs a backdated version and I
cannot seem to get it to install the older version

g9631519, 2024-12-03:
Same issue on Mac.
downloaded the package manually and fixed.