qiskit-community/qiskit-metal #993: Error in Tutorial 4 Analysis 4.04 New LOM and Fluxonium Example

auratrifire, 2024-07-02:
### Information

- Qiskit Metal **0.1.5**:
- Python **3.10.2**:
- **Windows 10**:

### What is the current behavior?
The running the line:
hamiltonian_results = composite_sys.hamiltonian_results(hilbertspace, evals_count=30)

results in an error - TypeError: must be real number, not Qobj

Full error below: 

Cell In[17], line 1
----> 1 hamiltonian_results = composite_sys.hamiltonian_results(hilbertspace, evals_count=30)

File c:\users\aurat\qiskit-metal\qiskit_metal\analyses\quantization\lom_core_analysis.py:1522, in CompositeSystem.hamiltonian_results(self, hilbertspace, evals_count, print_info)
   1519 esys_array[0] = evals
   1520 esys_array[1] = evecs
-> 1522 f01s, chi_mat = extract_energies(esys_array,
   1523                                  mode_size=hamiltonian_mat.dims[0])
   1524 f01s = f01s / 1000
   1525 ham_res['fQ_in_Ghz'] = dict(zip(names, f01s))

File c:\users\aurat\qiskit-metal\qiskit_metal\analyses\hamiltonian\states_energies.py:120, in extract_energies(esys_array, mode_size, zero_evals)
    111 double_excitation_states_mat = np.squeeze(
    112     np.array([
    113         double_excitation_states[ii]
    114         for ii in range(len(double_excitation_states))
    115     ])).T
    117 # Find the inner product of each of the target state with each of the
    118 # eigenvector; hence overlap has dimension of number of eigenvectors x number of target states
    119 overlap_single = np.absolute(
--> 120     np.array((Qobj(evecs_dag_mat) * Qobj(single_excitation_states_mat))))
    121 overlap_double = np.absolute(
    122     np.array((Qobj(evecs_dag_mat) * Qobj(double_excitation_states_mat))))
    124 # find the index of the eigenvector that is closest to each target state
    125 # hence evec_idx has shape of (number of target states, )

File [~\anaconda3\envs\qubit\lib\site-packages\qutip\core\qobj.py:307](http://localhost:8890/lab/tree/qiskit-metal/tutorials/4%20Analysis/A.%20Core%20-%20EM%20and%20quantization/~/anaconda3/envs/qubit/lib/site-packages/qutip/core/qobj.py#line=306), in Qobj.__init__(self, arg, dims, copy, superrep, isherm, isunitary)
    305 self._isherm = isherm
    306 self._isunitary = isunitary
--> 307 self._initialize_data(arg, dims, copy)
    309 if superrep is not None:
    310     self.superrep = superrep

File [~\anaconda3\envs\qubit\lib\site-packages\qutip\core\qobj.py:295](http://localhost:8890/lab/tree/qiskit-metal/tutorials/4%20Analysis/A.%20Core%20-%20EM%20and%20quantization/~/anaconda3/envs/qubit/lib/site-packages/qutip/core/qobj.py#line=294), in Qobj._initialize_data(self, arg, dims, copy)
    293         self._isunitary = arg._isunitary
    294 else:
--> 295     self._data = _data.create(arg, copy=copy)
    296     self._dims = Dimensions(
    297         dims or [[self._data.shape[0]], [self._data.shape[1]]]
    298     )
    299 if self._dims.shape != self._data.shape:

File [~\anaconda3\envs\qubit\lib\site-packages\qutip\core\data\convert.pyx:477](http://localhost:8890/lab/tree/qiskit-metal/tutorials/4%20Analysis/A.%20Core%20-%20EM%20and%20quantization/~/anaconda3/envs/qubit/lib/site-packages/qutip/core/data/convert.pyx#line=476), in qutip.core.data.convert._create.__call__()

File [~\anaconda3\envs\qubit\lib\site-packages\qutip\core\data\dense.pyx:43](http://localhost:8890/lab/tree/qiskit-metal/tutorials/4%20Analysis/A.%20Core%20-%20EM%20and%20quantization/~/anaconda3/envs/qubit/lib/site-packages/qutip/core/data/dense.pyx#line=42), in qutip.core.data.dense.Dense.__init__()

### Steps to reproduce the problem
Run the code up until part 6 of tutorial 4.04 of Qiskit Metal. 

### What is the expected behavior?
Give the results of the analyzed Fluxonium hamiltonian. 

### Suggested solutions
try except statement. 

Seems to be some error interfacing with casting an array as a Qobj when it is a scalar. 

the following would maybe work, but the solution may not be very general, and try, except statements usually aren't good code: 

try: 
    overlap_single = np.absolute(
        np.array((Qobj(evecs_dag_mat) * Qobj(single_excitation_states_mat))))
    overlap_double = np.absolute(
        np.array((Qobj(evecs_dag_mat) * Qobj(double_excitation_states_mat))))

except TypeError:
    overlap_single = np.absolute(
        np.array((evecs_dag_mat * Qobj(single_excitation_states_mat))))
    overlap_double = np.absolute(
        np.array((evecs_dag_mat * Qobj(double_excitation_states_mat))))

auratrifire, 2024-07-02:
I think I solved the bug,

I replaced

111 double_excitation_states_mat = np.squeeze(
112 np.array([
113 double_excitation_states[ii]
114 for ii in range(len(double_excitation_states))
115 ])).T
117 # Find the inner product of each of the target state with each of the
118 # eigenvector; hence overlap has dimension of number of eigenvectors x number of target states
119 overlap_single = np.absolute(
--> 120 np.array((Qobj(evecs_dag_mat) * Qobj(single_excitation_states_mat))))
121 overlap_double = np.absolute(
122 np.array((Qobj(evecs_dag_mat) * Qobj(double_excitation_states_mat))))

with: 

    # Find the inner product of each of the target state with each of the
    # eigenvector; hence overlap has dimension of number of eigenvectors x number of target states

    overlap_single = []
    overlap_double = []
    for u in evecs_dag_mat:
        for v in single_excitation_states_mat:
            overlap_single.append(u*v)
        for w in double_excitation_states_mat:
            overlap_double.append(u*w)

    overlap_single = np.reshape(np.absolute(
        np.array(overlap_single)), (*evecs_dag_mat.shape, *single_excitation_states_mat.shape))
    overlap_double = np.reshape(np.absolute(
        np.array(overlap_double)), (*evecs_dag_mat.shape, *double_excitation_states_mat.shape))

auratrifire, 2024-07-02:
Is there a way I can push my changes to the github?

closed 2024-07-02 by auratrifire

zlatko-minev, 2024-07-03:
Did you try making a pull request?