qiskit-community/qiskit-metal #994: Error in the interface between qiskit metal and ansys

SimonaZaccaria, 2024-07-09:
### Information

- **Qiskit Metal version**:0.1.5
- **Python version**:3.11.9
- **Operating system**: Windows

### What is the current behavior?
When I run the code in order to do epr analysis it starts to run but a certain time it stops and it gives me this error "com_error: (-2147352567, 'Eccezione.', (0, None, None, None, 0, -2147024382), None)", on ansys message windows I see "Script macro error: Invalid PropServer 'Freq. vs. pass:re(Mode(1)):Curve1'. (08:09:21 PM  Jul 09, 2024)"

### Steps to reproduce the problem
You can simple run one of the example at this page https://qiskit-community.github.io/qiskit-metal/tut/4-Analysis/4.02-Eigenmode-and-EPR.html# and when you arrive to the part of epr simulation you should see the error

### What is the expected behavior?
In principle it should gives me the value of epr and chi matrix, instead it starts to write the values and the stop with an error

### Suggested solutions

zlatko-minev, 2024-07-15:
What step is it at?

Xander-Ault, 2024-07-19:
I get a similar error when trying to use `run_epr` with `no_junctions=False`. The I get the following output:

```
INFO 09:26AM [connect_project]: Connecting to Ansys Desktop API...
INFO 09:26AM [load_ansys_project]:      Opened Ansys App
INFO 09:26AM [load_ansys_project]:      Opened Ansys Desktop v2024.2.0
INFO 09:26AM [load_ansys_project]:      Opened Ansys Project
        Folder:    C:/Users/xra/Documents/Ansoft/
        Project:   Project11
INFO 09:26AM [connect_design]:  Opened active design
        Design:    TransmonResonator_hfss [Solution type: Eigenmode]
INFO 09:26AM [get_setup]:       Opened setup `Setup`  (<class 'pyEPR.ansys.HfssEMSetup'>)
INFO 09:26AM [connect]:         Connected to project "Project11" and design "TransmonResonator_hfss" 😀

INFO 09:26AM [connect_design]:  Opened active design
        Design:    TransmonResonator_hfss [Solution type: Eigenmode]
INFO 09:27AM [get_setup]:       Opened setup `Setup`  (<class 'pyEPR.ansys.HfssEMSetup'>)
INFO 09:27AM [analyze]: Analyzing setup Setup
09:30AM 12s INFO [get_f_convergence]: Saved convergences to C:\Users\xra\qiskit-metal\test\hfss_eig_f_convergence.csv
INFO 09:30AM [get_setup]:       Opened setup `Setup`  (<class 'pyEPR.ansys.HfssEMSetup'>)
Design "TransmonResonator_hfss" info:
        # eigenmodes    2
        # variations    1
Design "TransmonResonator_hfss" info:
        # eigenmodes    2
        # variations    1

        energy_elec_all       = 7.41814629691302e-24
        energy_elec_substrate = 6.82309684535228e-24
        EPR of substrate = 92.0%

        energy_mag    = 6.45361541921198e-26
        energy_mag % of energy_elec_all  = 0.9%

Variation 0  [1/1]

  ←[1mMode 0 at 6.24 GHz   [1/2]←[0m
    Calculating ℰ_magnetic,ℰ_electric
       (ℰ_E-ℰ_H)/ℰ_E       ℰ_E       ℰ_H
               99.1%  3.709e-24 3.227e-26

    Calculating junction energy participation ration (EPR)
        method=`line_voltage`. First estimates:
        junction        EPR p_0j   sign s_0j    (p_capacitive)
                Energy fraction (Lj over Lj&Cj)= 97.02%
        jj              0.990632  (+)        0.0304596
                (U_tot_cap-U_tot_ind)/mean=1.53%
Calculating Qdielectric_main for mode 0 (0/1)
p_dielectric_main_0 = 0.9197846162985001

  ←[1mMode 1 at 9.32 GHz   [2/2]←[0m
    Calculating ℰ_magnetic,ℰ_electric
       (ℰ_E-ℰ_H)/ℰ_E       ℰ_E       ℰ_H
                0.3%  1.611e-24 1.606e-24

    Calculating junction energy participation ration (EPR)
        method=`line_voltage`. First estimates:
        junction        EPR p_1j   sign s_1j    (p_capacitive)
                Energy fraction (Lj over Lj&Cj)= 93.58%
        jj              0.00308315  (+)        0.000211468
                (U_tot_cap-U_tot_ind)/mean=0.01%
Calculating Qdielectric_main for mode 1 (1/1)
p_dielectric_main_1 = 0.9150894919855215
Traceback (most recent call last):
  File "C:\Users\xra\qiskit-metal\test\ansys_eigen_EPR_analysis_combo.py", line 63, in <module>
    eig_qres.run_epr()
  File "c:\users\xra\qiskit-metal\qiskit_metal\analyses\quantization\energy_participation_ratio.py", line 172, in run_epr
    self.run_analysis()
  File "c:\users\xra\qiskit-metal\qiskit_metal\analyses\quantization\energy_participation_ratio.py", line 219, in run_analysis
    self.sim.renderer.epr_run_analysis()
  File "c:\users\xra\qiskit-metal\qiskit_metal\renderers\renderer_ansys\ansys_renderer.py", line 1759, in epr_run_analysis
    self.epr_distributed_analysis.do_EPR_analysis()
  File "C:\Users\xra\AppData\Local\anaconda3\envs\qiskit\Lib\site-packages\pyEPR\core_distributed_analysis.py", line 1326, in do_EPR_analysis
    self._update_results(variation, Om, Pm, Sm, Qm_coupling, SOL,
  File "C:\Users\xra\AppData\Local\anaconda3\envs\qiskit\Lib\site-packages\pyEPR\core_distributed_analysis.py", line 1376, in _update_results
    self.results[variation]['convergence_f_pass'] = self.hfss_report_f_convergence(
                                                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\xra\AppData\Local\anaconda3\envs\qiskit\Lib\site-packages\pyEPR\core_distributed_analysis.py", line 1582, in hfss_report_f_convergence
    set_property(report, 'Attributes', curves, 'Line Width', 3)
  File "C:\Users\xra\AppData\Local\anaconda3\envs\qiskit\Lib\site-packages\pyEPR\ansys.py", line 357, in set_property
    return prop_holder.ChangeProperty([
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<COMObject GetModule>", line 2, in ChangeProperty
pywintypes.com_error: (-2147352567, 'Exception occurred.', (0, None, None, None, 0, -2147024382), None)
```

The error occurs in each of the tutorials in the Eigenmode and EPR analysis section of the qiskit-metal docs, but no such error  occurs if we change `no_junctions=True`

KPiran2007, 2024-07-23:
This error occurs both in the case of no_junctions=True and False and so is not a solution to the issue.

SimonaZaccaria, 2024-07-23:
> What step is it at?

run_epr()

I solved following a comment in the slack channel that suggested me to comment some line of code

Xander-Ault, 2024-07-23:
> I solved following a comment in the slack channel that suggested me to comment some line of code

Do you know which line it was? Or are you able to find the slack message again? This is an error that propagates to many of the following tutorials and a quick fix would be nice to move forward with for now until we can get it fixed in the code.

Xander-Ault, 2024-07-23:
> [zlatko-minev/pyEPR#142](https://github.com/zlatko-minev/pyEPR/pull/142) this is the link, there it's written to comment line from 1579 to 1587 in core_distributed_analysis.py (be carefull because in my file the code was shifted of a couple of lines)

This fixes it, thank you. 
For others trying to fix this, I recommend installing the correct version (0.9.0) of pyEPR-quantum before making the changes or you might run into this issue here [https://github.com/qiskit-community/qiskit-metal/issues/988](url).

Aadii-tya, 2025-09-10:
Hii, I'm facing the same issue, is it because of Ansys Student version doesnt have some functionalities or is it something else ?

zlatko-minev, 2025-09-10:
> Thankyou for the suggestion. I'll look into it.
> I was trying to simulate a Transmon based qubit.

For palace, there is an integration with metal called SQDMetal, https://github.com/sqdlab/SQDMetalWhich also works with squadds We would like to bring this natively into this repository here. That’s on the due for the future list of the community working group and then future maintainers.Sent from my iPhone

abhishekchak52, 2026-01-19:
> This fixes it, thank you. For others trying to fix this, I recommend installing the correct version (0.9.0) of pyEPR-quantum before making the changes or you might run into this issue here [...]

Sounds like this was fixed in a previous release. Closing as completed. The Ansys interface is due for a refresh later in 2026. We can open another issue for problems related to the new interface.