qiskit-community/qiskit-metal #995: Junction not being render in Ansys

Risav25Pokhrel, 2024-07-21:
### Information

- **Qiskit Metal version**: 0.1.5
- **Python version**:3.9.19
- **Operating system**:Windows 11

### What is the current behavior?

Cell In[18], [line 1](vscode-notebook-cell:?execution_count=18&line=1)
----> [1](vscode-notebook-cell:?execution_count=18&line=1) eig_qb.sim.run_sim(name="Qbit")

File c:\Users\USER\anaconda3\envs\qmetal\lib\site-packages\qiskit_metal\analyses\simulation\eigenmode.py:130, in EigenmodeSim.run_sim(self, name, components, open_terminations, port_list, jj_to_port, ignored_jjs, box_plus_buffer)
    [127](file:///C:/Users/USER/anaconda3/envs/qmetal/lib/site-packages/qiskit_metal/analyses/simulation/eigenmode.py:127)     self._initialize_renderer()
    [129](file:///C:/Users/USER/anaconda3/envs/qmetal/lib/site-packages/qiskit_metal/analyses/simulation/eigenmode.py:129) vars_to_initialize = self.setup.vars
--> [130](file:///C:/Users/USER/anaconda3/envs/qmetal/lib/site-packages/qiskit_metal/analyses/simulation/eigenmode.py:130) renderer_design_name = self._render(
    [131](file:///C:/Users/USER/anaconda3/envs/qmetal/lib/site-packages/qiskit_metal/analyses/simulation/eigenmode.py:131)     name=name,
    [132](file:///C:/Users/USER/anaconda3/envs/qmetal/lib/site-packages/qiskit_metal/analyses/simulation/eigenmode.py:132)     solution_type='eigenmode',
    [133](file:///C:/Users/USER/anaconda3/envs/qmetal/lib/site-packages/qiskit_metal/analyses/simulation/eigenmode.py:133)     selection=components,
    [134](file:///C:/Users/USER/anaconda3/envs/qmetal/lib/site-packages/qiskit_metal/analyses/simulation/eigenmode.py:134)     open_pins=open_terminations,
    [135](file:///C:/Users/USER/anaconda3/envs/qmetal/lib/site-packages/qiskit_metal/analyses/simulation/eigenmode.py:135)     port_list=port_list,
    [136](file:///C:/Users/USER/anaconda3/envs/qmetal/lib/site-packages/qiskit_metal/analyses/simulation/eigenmode.py:136)     jj_to_port=jj_to_port,
    [137](file:///C:/Users/USER/anaconda3/envs/qmetal/lib/site-packages/qiskit_metal/analyses/simulation/eigenmode.py:137)     ignored_jjs=ignored_jjs,
    [138](file:///C:/Users/USER/anaconda3/envs/qmetal/lib/site-packages/qiskit_metal/analyses/simulation/eigenmode.py:138)     box_plus_buffer=box_plus_buffer,
    [139](file:///C:/Users/USER/anaconda3/envs/qmetal/lib/site-packages/qiskit_metal/analyses/simulation/eigenmode.py:139)     vars_to_initialize=vars_to_initialize)
    [141](file:///C:/Users/USER/anaconda3/envs/qmetal/lib/site-packages/qiskit_metal/analyses/simulation/eigenmode.py:141) self._analyze()
    [142](file:///C:/Users/USER/anaconda3/envs/qmetal/lib/site-packages/qiskit_metal/analyses/simulation/eigenmode.py:142) return renderer_design_name, self.sim_setup_name

File c:\Users\USER\anaconda3\envs\qmetal\lib\site-packages\qiskit_metal\analyses\core\simulation.py:167, in QSimulation._render(self, solution_type, vars_to_initialize, **design_selection)
    [165](file:///C:/Users/USER/anaconda3/envs/qmetal/lib/site-packages/qiskit_metal/analyses/core/simulation.py:165)     del design_selection["name"]
    [166](file:///C:/Users/USER/anaconda3/envs/qmetal/lib/site-packages/qiskit_metal/analyses/core/simulation.py:166) design_name = base_name + "_" + self.renderer_name
--> [167](file:///C:/Users/USER/anaconda3/envs/qmetal/lib/site-packages/qiskit_metal/analyses/core/simulation.py:167) design_name = self.renderer.execute_design(
...
    [208](file:///C:/Users/USER/anaconda3/envs/qmetal/lib/re.py:208)     a callable, it's passed the Match object and must return
    [209](file:///C:/Users/USER/anaconda3/envs/qmetal/lib/re.py:209)     a replacement string to be used."""
--> [210](file:///C:/Users/USER/anaconda3/envs/qmetal/lib/re.py:210)     return _compile(pattern, flags).sub(repl, string, count)

TypeError: expected string or bytes-like object

### Steps to reproduce the problem
I have made a custom Component and is working well in design but while doing epr analysis it throws this error
[image]

In hfss the junction is not being rendered
[image]

zlatko-minev, 2026-09-27:
Thanks for the report, and sorry for the slow response.

The `re.sub` error during `execute_design` most likely comes from the renderer's name sanitiser (`get_clean_name`), which assumes every qgeometry `name` is a string. Those names come from the dictionary keys you pass to `self.add_qgeometry(...)` in your component's `make()`. On current `main` we can reproduce this exact TypeError with a custom component that adds its junction under a non-string key, e.g. `self.add_qgeometry('junction', {0: jj_line}, width=...)`. Junctions are rendered after paths and polygons, which would also explain why the pads appear in HFSS but the junction doesn't.

#1187 makes `add_qgeometry` convert such keys to strings, so this can't happen regardless of the key type. We still can't confirm it's your case, because the traceback is truncated at `...` and we can't see the component code. Could you post the full traceback and the `add_qgeometry` calls from your `make()`? In the meantime, a string name such as `dict(rect_jj=jj_line)` should unblock you.

---
_Generated by [Claude Code](https://claude.ai/code)_

zlatko-minev, 2026-11-15:
Thanks @Risav25Pokhrel for the report. The `re.sub` TypeError comes from a geometry
name that isn't a string, such as `self.add_qgeometry('junction', {0: jj_line}, ...)`. Junctions are
rendered after pads, which would explain pads appearing without the junction. Since
v0.9.0 (#1187) `add_qgeometry` converts such keys to strings, so this TypeError can no
longer happen. We never saw your full traceback, so if you still see a missing junction
on a current release, please reopen with the traceback and the `add_qgeometry` calls from
your `make()`. Closing for now.

closed 2026-11-15 by zlatko-minev