qiskit-community/qiskit-metal #996: DrivenModal Solution Type not Supported in Newer Version of Ansys

Xander-Ault, 2024-07-22:
### Information

- **Qiskit Metal version**:  0.1.5
- **Python version**:  3.11.9
- **Operating system**:  Windows 10
- **Ansys version**: 2024 R2

### What is the current behavior?
When trying to run the Impedance, admittance and scattering analysis tutorial on the qiskit-metal docs, an error occurs while using newer versions of Ansys. The error occurs when running the command ```em1.run( name="SingleTM_hfss", components=[], open_terminations=[], port_list=[('Q1', 'readout', 70)], jj_to_port=[('Q1', 'rect_jj', 50, True)], box_plus_buffer = False)```. The first error is as follows: 
```
Traceback (most recent call last):
  File "C:\Users\xra\AppData\Local\anaconda3\envs\qiskit\Lib\site-packages\pyEPR\project_info.py", line 335, in connect_setup
    self.setup_name = setup.name
                      ^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'name'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\xra\qiskit-metal\test\DO_NOT_RUN.py", line 46, in <module>
    em1.run( name="SingleTM_hfss", components=[], open_terminations=[], port_list=[('Q1', 'readout', 70)],
  File "c:\users\xra\qiskit-metal\qiskit_metal\analyses\core\simulation.py", line 203, in run
    self.run_sim(*args, **kwargs)
  File "c:\users\xra\qiskit-metal\qiskit_metal\analyses\simulation\scattering_impedance.py", line 175, in run_sim
    renderer_design_name = self._render(
                           ^^^^^^^^^^^^^
  File "c:\users\xra\qiskit-metal\qiskit_metal\analyses\core\simulation.py", line 167, in _render
    design_name = self.renderer.execute_design(
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "c:\users\xra\qiskit-metal\qiskit_metal\renderers\renderer_ansys\ansys_renderer.py", line 753, in execute_design
    self.new_ansys_design(design_name, solution_type)
  File "c:\users\xra\qiskit-metal\qiskit_metal\renderers\renderer_ansys\ansys_renderer.py", line 795, in new_ansys_design
    self.connect_ansys_design(adesign.name)
  File "c:\users\xra\qiskit-metal\qiskit_metal\renderers\renderer_ansys\ansys_renderer.py", line 463, in connect_ansys_design
    self.pinfo.connect_setup()
  File "C:\Users\xra\AppData\Local\anaconda3\envs\qiskit\Lib\site-packages\pyEPR\project_info.py", line 346, in connect_setup
    raise Exception(' Did you provide the correct setup name?\
  File "C:\Users\xra\AppData\Local\anaconda3\envs\qiskit\Lib\site-packages\pyEPR\project_info.py", line 335, in connect_setup
    self.setup_name = setup.name
                      ^^^^^^^^^^
Exception:  Did you provide the correct setup name?                            Failed to pull up setup. 😭
```
I think this error is an easy fix and merely a problem of case usage in the code as described below. However, once I fix this error using my attempted solution, I get the following error instead
```
11:20AM 09s ERROR [new_ansys_design]: The solution_type = DrivenModal is not supported by this renderer
Traceback (most recent call last):
  File "C:\Users\xra\qiskit-metal\test\DO_NOT_RUN.py", line 46, in <module>
    em1.run( name="SingleTM_hfss", components=[], open_terminations=[], port_list=[('Q1', 'readout', 70)],
  File "c:\users\xra\qiskit-metal\qiskit_metal\analyses\core\simulation.py", line 203, in run
    self.run_sim(*args, **kwargs)
  File "c:\users\xra\qiskit-metal\qiskit_metal\analyses\simulation\scattering_impedance.py", line 175, in run_sim
    renderer_design_name = self._render(
                           ^^^^^^^^^^^^^
  File "c:\users\xra\qiskit-metal\qiskit_metal\analyses\core\simulation.py", line 167, in _render
    design_name = self.renderer.execute_design(
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "c:\users\xra\qiskit-metal\qiskit_metal\renderers\renderer_ansys\ansys_renderer.py", line 753, in execute_design
    self.new_ansys_design(design_name, solution_type)
  File "c:\users\xra\qiskit-metal\qiskit_metal\renderers\renderer_ansys\ansys_renderer.py", line 795, in new_ansys_design
    self.connect_ansys_design(adesign.name)
                              ^^^^^^^
UnboundLocalError: cannot access local variable 'adesign' where it is not associated with a value
```

### Steps to reproduce the problem

Run the code found here [https://qiskit-community.github.io/qiskit-metal/tut/4-Analysis/4.03-Impedance.html](url) (the Impedance, admittance and scattering analysis tutorial available in the qiskit-metal docs) using Ansys Electronics Desktop 2024 R2. 

### What is the expected behavior?

The expect output is as it is in the tutorial:
```
INFO 08:50AM [connect_project]: Connecting to Ansys Desktop API...
INFO 08:50AM [load_ansys_project]:      Opened Ansys App
INFO 08:50AM [load_ansys_project]:      Opened Ansys Desktop v2020.2.0
INFO 08:50AM [load_ansys_project]:      Opened Ansys Project
        Folder:    C:/Ansoft/
        Project:   Project22
INFO 08:50AM [connect_design]: No active design found (or error getting active design).
INFO 08:50AM [connect]:          Connected to project "Project22". No design detected
INFO 08:50AM [connect_design]:  Opened active design
        Design:    SingleTM_hfss [Solution type: DrivenModal]
WARNING 08:50AM [connect_setup]:        No design setup detected.
WARNING 08:50AM [connect_setup]:        Creating drivenmodal default setup.
INFO 08:50AM [get_setup]:       Opened setup `Setup`  (<class 'pyEPR.ansys.HfssDMSetup'>)
INFO 08:50AM [get_setup]:       Opened setup `Setup1`  (<class 'pyEPR.ansys.HfssDMSetup'>)
INFO 08:50AM [get_setup]:       Opened setup `Setup1`  (<class 'pyEPR.ansys.HfssDMSetup'>)
INFO 08:50AM [analyze]: Analyzing setup Setup1 : Sweep
```

### Suggested solutions

The solve the first error I described above, it seems to be a simple misuse of case in the code. In `qiskit_metal\analyses\simulation\scattering_impedance.py` on line 177, there is `solution_type='drivenmodal'`, but in `pyEPR\project_info.py` it is checking `elif self.design.solution_type == 'DrivenModal':` on line 326. Changing the case in `scattering_impedance.py` to match that in `project_info.py` then produces the second error described above. This may be another case of a problem with variable names as in `qiskit_metal\renderers\renderer_ansys\ansys_renderer.py` on line 780 there is `elif solution_type == "drivenmodal":`, although changing all 3 of these to have the same case produces the original error. Another possible explanation is that DrivenModal is no longer part of the Ansys 2024 R2 suite. It seems that they have changed the solution type name from DrivenModal to ModalNetwork in newer releases of the software.

SssZKing, 2024-12-06:
You're right! I believe the name change is the reason for this problem. What I did to manage(or bypass) it is to add a few lines corresponding to the `DrivenModal` and I have to manually change the solution type in HFSS from `Hybrid Modal` to `Modal Network` after creating a new design. The following are the detailed locations:

`C:\Users\zshen\qiskit-metal\qiskit_metal\renderers\renderer_ansys\ansys_renderer.py -> new_ansys_design
	elif solution_type == "hfss modal network":
		adesign = self.pinfo.project.new_dm_design(design_name)

C:\Users\zshen\qiskit-metal\qiskit_metal\renderers\renderer_ansys\ansys_renderer.py -> new_ansys_setup
	elif self.pinfo.design.solution_type == "HFSS Modal Network":
		setup = self.add_drivenmodal_setup(name, **other_setup)

C:\Users\zshen\.conda\envs\qiskit_env\Lib\site-packages\pyEPR\ansys.py -> get_setup
	elif self.solution_type == "HFSS Modal Network":
		return HfssDMSetup(self, name)`

sirshisram, 2025-01-20:
Hey @SssZKing I too am having the same problem. I did what you had suggested, and I'm still facing the same issue.

SssZKing, 2025-01-22:
Hi @sirshisram. I am running on 2022 R2, but I'm not sure whether the names of solution types changed further. I guess you should edit names accordingly. Also, maybe double check the path, it could related to how you installed qiskit-metal and its Python environment.

zlatko-minev, 2026-11-15:
Thanks @Xander-Ault for the report and the traceback, @SssZKing for the workaround, and
@PositroniumJS for confirming the fix. HFSS 2024.1+ renamed Driven Modal to
"HFSS Modal Network" / "HFSS Hybrid Modal Network". pyEPR handles the new names since
pyEPR PR #176, and quantum-metal accepts them since v0.6.0 (#1051). Current releases require
`pyEPR-quantum>=1.0.2`, so `pip install --upgrade "quantum-metal[ansys]"` picks this up.
Closing.

closed 2026-11-15 by zlatko-minev