qiskit-community/qiskit-metal #998: Error in 'run_epr' step in qiskit metal with ansys2023

Shawn-Seo93, 2024-09-12:
### Information

- **Qiskit Metal version**:  0.1.5
- **Python version**:  3.9.19
- **Operating system**:  Window

### What is the current behavior?
When runing tutorial code 'https://qiskit-community.github.io/qiskit-metal/tut/4-Analysis/4.02-Eigenmode-and-EPR.html',
I got an error message : "com_error: (-2147352567, 'Eccezione.', (0, None, None, None, 0, -2147024382), None)"

I'm using Ansys2023, Qiskit Metal 0.1.5, python version 3.9.19

I tried the suggested solution in 'https://github.com/zlatko-minev/pyEPR/pull/142'
but after I replace the previous 1579-1587 line, New error by 'if' occurs.

### Steps to reproduce the problem
You can simple run one of the example at this page https://qiskit-community.github.io/qiskit-metal/tut/4-Analysis/4.02-Eigenmode-and-EPR.html# and when you arrive to the part of epr simulation you should see the error

### What is the expected behavior?
In principle it should gives me the value of epr and chi matrix, instead it starts to write the values and the stop with an error

### Suggested solutions

zlatko-minev, 2024-09-12:
Can you post the error trace? It’s not clear what the error is

I tried the suggested solution in 'zlatko-minev/pyEPR#142'

Shawn-Seo93, 2024-09-12:
Sorry, I have tried many possible combination of software versions, and found it finally working!
My problem was solved in the conditions below.
Maybe it was caused by installation version mixing.,,

Qiskit Metal version:  0.1.5
Python version:  3.9.19
Ansys 2021R1
numpy version : 1.23.2
Operating system:  Window

zlatko-minev, 2024-09-12:
Great!

Nour-Aldein2, 2024-10-04:
I'm encountering a similar issue while using Ansys 2023R2. I've followed the setup as outlined with the same package versions except for Ansys, which is the updated 2023R2 version. Below is the error message I'm receiving when attempting to run the `eig_qb.run_epr()` function:

`com_error                                 Traceback (most recent call last)
Cell In[17], line 1
...
com_error: (-2147352567, 'Exception occurred.', (0, None, None, None, 0, -2147024382), None)
`

Is there any known incompatibility with Ansys 2023R2, or any suggestions for debugging this issue further?

zlatko-minev, 2026-09-27:
Glad you got it working, @Shawn-Seo93, and thanks for posting the combination that worked for you.

A lot has changed since this was filed: the package is now `quantum-metal`, and the Ansys extra requires `pyEPR-quantum>=1.0.1`. There's no traceback in this thread, so we can't tell whether the original Ansys 2023 error still happens on current releases, and we can't reproduce it without Ansys.

I'll close this as resolved. If anyone hits a `com_error` during `run_epr` on a current release (`pip install -U "quantum-metal[ansys]"`), please open a new issue with the full traceback, your Ansys version, and the output of `pip show quantum-metal pyEPR-quantum numpy`.

---
_Generated by [Claude Code](https://claude.ai/code)_

closed 2026-09-27 by zlatko-minev