qiskit-community/qiskit-metal #999: Installation from source on linux - process killed

draganal28, 2024-10-22:
### Information

- **Qiskit Metal version - latest from the source as per 22.10.2024**:
- **Python version - the one that comes along as specified in environment.yml**:
- **Operating system - linux: Linux ThinkPad-X1-Carbon-2nd 5.19.0-46-generic #47-Ubuntu SMP PREEMPT_DYNAMIC Fri Jun 16 13:30:11 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux**:

### What is the current behavior?
I follow recommended installation from source (Option 1, create new environment) but each time I try it (and I have been trying for the last three days), there is the same outcome:
dragana@ThinkPad-X1-Carbon-2nd:~/qiskit-workspace/workshop_SCQHW_fall_2024/qiskit-metal$ conda env create -n qiskitmetal -f environment.yml
Collecting package metadata (repodata.json): done
Solving environment: | Killed

(attached option_1_install_new_environment.png screnshot)

dmesg:
[ 2810.246062] oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),cpuset=/,mems_allowed=0,global_oom,task_memcg=/user.slice/user-1000.slice/user@1000.service/app.slice/app-org.gnome.Terminal.slice/vte-spawn-8c6f265e-a98d-43cd-8cef-bb86c7f858fe.scope,task=conda-env,pid=6308,uid=1000
[ 2810.246093] Out of memory: Killed process 6308 (conda-env) total-vm:6556824kB, anon-rss:5371692kB, file-rss:0kB, shmem-rss:0kB, UID:1000 pgtables:12164kB oom_score_adj:200
[ 2810.257458] Purging GPU memory, 0 pages freed, 0 pages still pinned, 5440 pages left available.

(attached dmesg_1.png screnshot)

However, the memory does not seem to be that low:
dragana@ThinkPad-X1-Carbon-2nd:~/qiskit-workspace/workshop_SCQHW_fall_2024/qiskit-metal$ free
               total        used        free      shared  buff/cache   available
Mem:         7761456     2034620     4449856      412128     1276980     4986912
Swap:        2097148     1167976      929172

(attached memory_snapshot.png screenshot)

I am experienced with working with conda environments and installation of sw and I have never experienced a case like this. 
I also tried creating a new clean environment then installing one by one package from environment.yml but at fourth the same kind of kill process happens (oom). 

I have succeeded in installing qiskit-metal on macos by using the same commands.

### Steps to reproduce the problem
git clone https://github.com/Qiskit/qiskit-metal.git
cd qiskit-metal
conda env create -n qiskitmetal -f environment.yml

### What is the expected behavior?
It is expected to have new conda environment named qiskitmetal (in my case the name I gave to new environment) created.

### Suggested solutions
I would suggest the team inspect how installation process is realised regarding depndencies as that it is the only thing I could think of which might cause OOM situation. Although it seems very unlikely to need that much memory. Can it be some recurrent link or circular link chain?

zlatko-minev, 2024-10-22:
Sounds like something very specific to your Lennox environment. You could try the virtual environment set up instead of conda Sent from my iPhone

[ 2810.246062] ***@***.***/app.slice/app-org.gnome.Terminal.slice/vte-spawn-8c6f265e-a98d-43cd-8cef-bb86c7f858fe.scope,task=conda-env,pid=6308,uid=1000

draganal28, 2024-10-22:
So, you'd think it is linux environment. I can give it a try by increasing oom score from 200 (visible from dmesg output) to 1000 and run the same  command again. I have installation on macos, though, so I'll test it in a little while back on linux and if I find the cause for such behaviour, I'll share it. 

Thank you for a quick response.

zlatko-minev, 2026-09-27:
Thanks for the thorough report, dmesg and memory output included, @draganal28, and sorry this sat for so long.

The install story has changed quite a bit since then:
- Since v0.7.0 the default is a lightweight pip install: `pip install quantum-metal` (add `[gui]`, `[ansys]`, `[mesh]` or `[full]` as needed). No conda solve is involved.
- For a source install, the recommended route is now uv: `uv venv --python 3.11 && uv pip install -e .` (see "Installation from source" in `docs/installation.rst`).
- `environment.yml` no longer pins exact versions of ~20 packages, which is what the conda solver was working through when it was killed.

We haven't re-tested the conda path on a memory-limited Linux machine, so I can't promise that route is fixed. If `conda env create -f environment.yml` still gets OOM-killed on a current checkout, please reopen with your conda version. Newer conda releases default to the libmamba solver, which is usually much lighter on memory. Closing for now, since the pip/uv paths avoid the problem.

---
_Generated by [Claude Code](https://claude.ai/code)_

closed 2026-09-27 by zlatko-minev