qiskit-community/qiskit-metal #1000: Error when downloading through pip

Ethanooool, 2024-11-03:
### Information

- **Qiskit Metal version**: N/A
- **Python version**: 3.12
- **Operating system**: Windows

### What is the current behavior?
[image]

### Steps to reproduce the problem

when running pip install qiskit-metal

anishthomas9555, 2024-11-28:
Happens the same for Mac system as well

Ethanooool, 2024-11-28:
I was able to solve the issue with using conda and following the step for it

zlatko-minev, 2024-11-28:
I think this pull request should solve it 

Let us know in the PR

https://github.com/qiskit-community/qiskit-metal/pull/1002

PositroniumJS, 2024-12-20:
> I think this pull request should solve it
> 
> Let us know in the PR
> 
> #1002

It is not solving the problem. The error occurs when installing `numpy==1.24.2`.

This error occurs even on a bare new virtual env (Windows 11, Python 3.12.7)
```cmd
pip install -U setuptools
pip install numpy==1.24.2
Collecting numpy==1.24.2
  Using cached numpy-1.24.2.tar.gz (10.9 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error

  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [33 lines of output]
      Traceback (most recent call last):
        File "C:\Users\account\tempo\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 353, in <module>
          main()
        File "C:\Users\account\tempo\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\account\tempo\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 112, in get_requires_for_build_wheel
          backend = _build_backend()
                    ^^^^^^^^^^^^^^^^
        File "C:\Users\account\tempo\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 77, in _build_backend
          obj = import_module(mod_path)
                ^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Program Files\Python312\Lib\importlib\__init__.py", line 90, in import_module
          return _bootstrap._gcd_import(name[level:], package, level)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
        File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
        File "<frozen importlib._bootstrap>", line 1310, in _find_and_load_unlocked
        File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
        File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
        File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
        File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
        File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
        File "<frozen importlib._bootstrap_external>", line 995, in exec_module
        File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
        File "C:\Users\account\AppData\Local\Temp\pip-build-env-81_59slm\overlay\Lib\site-packages\setuptools\__init__.py", line 16, in <module>
          import setuptools.version
        File "C:\Users\account\AppData\Local\Temp\pip-build-env-81_59slm\overlay\Lib\site-packages\setuptools\version.py", line 1, in <module>
          import pkg_resources
        File "C:\Users\account\AppData\Local\Temp\pip-build-env-81_59slm\overlay\Lib\site-packages\pkg_resources\__init__.py", line 2172, in <module>
          register_finder(pkgutil.ImpImporter, find_on_path)
                          ^^^^^^^^^^^^^^^^^^^
      AttributeError: module 'pkgutil' has no attribute 'ImpImporter'. Did you mean: 'zipimporter'?
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
```

Bumping the version to `1.26.4` solves this numpy installation issue

anishthomas9555, 2024-12-22:
> > I think this pull request should solve it
> > Let us know in the PR
> > #1002
> 
> It is not solving the problem. The error occurs when installing `numpy==1.24.2`.
> 
> This error occurs even on a bare new virtual env (Windows 11, Python 3.12.7)
> 
> ```batchfile
> pip install -U setuptools
> pip install numpy==1.24.2
> Collecting numpy==1.24.2
>   Using cached numpy-1.24.2.tar.gz (10.9 MB)
>   Installing build dependencies ... done
>   Getting requirements to build wheel ... error
>   error: subprocess-exited-with-error
> 
>   × Getting requirements to build wheel did not run successfully.
>   │ exit code: 1
>   ╰─> [33 lines of output]
>       Traceback (most recent call last):
>         File "C:\Users\account\tempo\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 353, in <module>
>           main()
>         File "C:\Users\account\tempo\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 335, in main
>           json_out['return_val'] = hook(**hook_input['kwargs'])
>                                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>         File "C:\Users\account\tempo\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 112, in get_requires_for_build_wheel
>           backend = _build_backend()
>                     ^^^^^^^^^^^^^^^^
>         File "C:\Users\account\tempo\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 77, in _build_backend
>           obj = import_module(mod_path)
>                 ^^^^^^^^^^^^^^^^^^^^^^^
>         File "C:\Program Files\Python312\Lib\importlib\__init__.py", line 90, in import_module
>           return _bootstrap._gcd_import(name[level:], package, level)
>                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
>         File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
>         File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
>         File "<frozen importlib._bootstrap>", line 1310, in _find_and_load_unlocked
>         File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
>         File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
>         File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
>         File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
>         File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
>         File "<frozen importlib._bootstrap_external>", line 995, in exec_module
>         File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
>         File "C:\Users\account\AppData\Local\Temp\pip-build-env-81_59slm\overlay\Lib\site-packages\setuptools\__init__.py", line 16, in <module>
>           import setuptools.version
>         File "C:\Users\account\AppData\Local\Temp\pip-build-env-81_59slm\overlay\Lib\site-packages\setuptools\version.py", line 1, in <module>
>           import pkg_resources
>         File "C:\Users\account\AppData\Local\Temp\pip-build-env-81_59slm\overlay\Lib\site-packages\pkg_resources\__init__.py", line 2172, in <module>
>           register_finder(pkgutil.ImpImporter, find_on_path)
>                           ^^^^^^^^^^^^^^^^^^^
>       AttributeError: module 'pkgutil' has no attribute 'ImpImporter'. Did you mean: 'zipimporter'?
>       [end of output]
> 
>   note: This error originates from a subprocess, and is likely not a problem with pip.
> 
> × Getting requirements to build wheel did not run successfully.
> │ exit code: 1
> ╰─> See above for output.
> 
> note: This error originates from a subprocess, and is likely not a problem with pip.
> ```
> 
> Bumping the version to `1.26.4` solves this numpy installation issue

I tried that just now. Still the same error unfortunately

PositroniumJS, 2024-12-22:
> > Bumping the version to `1.26.4` solves this numpy installation issue
> 
> I tried that just now. Still the same error unfortunately

Depending on your python version, this numpy version should change, try first to install only numpy and edit qiskit metals's requirement file to use this version. Maybe you might face problem with other packages afterward.

Personnaly with python 3.12.7, I got this file:

```
addict==2.4.0
descartes==1.1.0
gdspy==1.6.13
geopandas==0.12.2
ipython==8.30.0
matplotlib==3.9.3
numpy==1.26.4
pandas==1.5.3
pint==0.20.1
pyEPR-quantum==0.9.0
pygments==2.18.0
pyside6
qdarkstyle==3.1
qutip==4.7.6
scipy==1.12.0
shapely==2.0.6
scqubits==3.1.0
gmsh==4.11.1
pyaedt==0.13.0
pyyaml==6.0.2
cython<3.0.0
```

but I can't tell you if everything is working well. Moreover, it should still be possible to bump some packages without breaking the code.

zlatko-minev, 2025-01-26:
Changed to a whole new set of requirements with less fixed versions, but lots of things need to be tested still in #1002

anishthomas9555, 2025-03-02:
Ok i think i finally got it working.
My mistake was that i kept trying to install all the requirements (pkgutil, different pip versions) on the base mac terminal. 

So i found this link https://gist.github.com/SamWolski/9f5bd115223a014f82ff8962cdbbb6b5#file-install_qm_m2-sh and created it on a new environment. After that it i was able to follow the transmon tutorial on VS code.

zlatko-minev, 2025-03-02:
Great! Sent from my iPhone

PositroniumJS, 2025-03-02:
> Ok i think i finally got it working. My mistake was that i kept trying to install all the requirements (pkgutil, different pip versions) on the base mac terminal.
>
> So i found this link [...]

This script works because on the first line it specifies to work with `python=3.10` so the package `numpy=1.24.2` is still compatible. See #1001

abhishekchak52, 2025-12-14:
Should be fixed now by #1002 and #1026. Please reopen if this issue persists.